RobotWebTools/rosbridge_suite #380: Problems in image type ros sensor message

wfcola, 2018-12-25:
The following code is the image node for picture publishing.   
    cv::Mat image = cv::imread(filename);
    sensor_msgs::ImagePtr msg = cv_bridge::CvImage(std_msgs::Header(), "bgr8", image).toImageMsg();
    pub.publish(msg);

However, when i subscribe the relating topic of this node from websocket test webpage, the length of the picture data is 1228800 which is different from the one published by image node of 921600. And it's seems that the data in every one quater is right and the other three quaters have problem like pixel setoff.

mvollrath, 2019-01-04:
It might be that what you're receiving on the client side is a base64 encoded string with the data in it, considering the length of data you're getting is exactly 4:3 of what you're expecting (this is how much bigger base64 encoding is than the original).  Try using compression "cbor" which gives you a `Uint8Array` instead of a base64 string, or you can try decoding the base64.

wfcola, 2019-01-04:
@mvollrath Thanks for your reply. This problem has been solved. It was cased by the coding principle. 
For example, the value of a pixel in bgr8 coding  is (123， 22， 210). Then the binary coding of this pixel is regarded as （0111 1011， 0001 0110， 1101 0010）. While transporting the data of the pixel, it is divided into 4 pieces by 6 bits of each. However, the data received on the Java script web page in websocket , is treated each piece in 8 bits and the data of the same pixel is changed into (0001 1110, 0011 0001,0001 1011, 0001 0010) . So, the following transform can remove the bits that should not be in the data.
 imgdata.data[m+2]=(arr1&0x3f) << 2 | (arr2&0x30) >> 4;//r
 imgdata.data[m+1]=(arr2&0x0f) << 4 | (arr3&0x3c) >> 2;//g
 imgdata.data[m]=(arr3&0x03) << 6 | (arr4&0x3f);//b
This works fine.

closed 2019-01-04 by wfcola