RobotWebTools/rosbridge_suite #394: BadYieldError: yielded unknown object None

skycoop, 2019-03-27:
Just updated to `rosbridge_server` v0.10.2 and I'm now being spammed with the following log message hundreds of times per second
```
[ERROR] [1553705564.030598]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_server/websocket_handler.py", line 168, in prewrite_message
    yield self.write_message(message, binary)
  File "/usr/local/lib/python2.7/dist-packages/tornado/gen.py", line 628, in run
    value = future.result()
  File "/usr/local/lib/python2.7/dist-packages/tornado/concurrent.py", line 111, in result
    raise self._exception
BadYieldError: yielded unknown object None
```

As far as I can tell, the immediate cause was the addition of exception logging in #386, but it looks like this exception was being thrown before then. As far as I can tell this exception has no actual impact on the functionality of the server, but after #386 produces a lot of noise.

mvollrath, 2019-03-27:
I see, this is happening in kinetic but not melodic.  Working on a fix.

skycoop, 2019-03-27:
Whoops. Probably would have been useful to mention that I was on kinetic. Thanks :smile:

closed 2019-03-27 by mvollrath

skycoop, 2019-03-27:
Thank you so much for that fix. Any ideas on when it might be released?

mvollrath, 2019-03-27:
The next kinetic sync may not be for a few weeks, I'll see if there are any other features we can merge quickly and then submit a release.

mvollrath, 2019-03-29:
0.11.0 has been submitted, watch the [ROS discourse](https://discourse.ros.org/c/release/kinetic) for the next sync.

PR #395 (2019-03-27): Handle BadYieldError in Tornado <4.5.0
Fixes #394