RobotWebTools/rosbridge_suite #399: Subscriber unable to receive messages

leviathan747, 2019-04-19:
## Behavior

I'm not sure if this is the expected behavior or not...

I have an application using roslibpy (https://github.com/gramaziokohler/roslibpy) to connect to a rosbridge server and communicate. It needs to publish and subscribe to topics. The flow goes something like this:

1. Create client, connect to rosbridge
2. Subscribe to some topics, publish to some topics, communicate back and forth with ROS
3. Disconnect from rosbridge and exit

It works the first time the client application runs, but the second time, it is able to publish messages but it is not able to receive messages it subscribes to. I was looking at the code and saw in `subscribers.py` line 217 that when a topic no longer has any subscribers, rosbridge unregisters the topic. I commented out these lines and my application now works.

Why is this behavior there? Is there a workaround for something like this? I cannot guarantee that there will always be at least one subscriber for the topic and I want to be able to connect a new client and subscribe to messages.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.0
  - Tornado Version (`python -c 'import tornado; print tornado.version'`): 4.5.3

gonzalocasas, 2019-04-25:
@leviathan747 I think you're bumping into a pretty old issue that has been mitigated but not totally resolved. The thread of the issue can be followed from this one: 
https ://github.com/RobotWebTools/rosbridge_suite/issues/239

The problem is basically how the reconnects are handled, internally there's a 10 second timeout to reconnect a client. The timeout was designed to deal with fast reconnects coming from `roslibjs` being a webpage and being reloaded often, but in `roslibpy` typical usage, your reconnection intervals will be much larger. The timeout can be configured (check the mitigation PR linked somewhere in that issue), so a larger value might be a temporary good-enough fix for you, but I believe a more sustainable solution should be proposed.

leviathan747, 2019-04-25:
@gonzalocasas yeah, I saw that thread and others. I actually had the same problem on the publish side and used the timeout configuration capability introduced by [this PR](https://github.com/RobotWebTools/rosbridge_suite/pull/322). But this problem has to do with topics being unregistered when there are no subscribers. For now, I just changed a few lines in my fork to make sure that they do not get unregistered (see [here](https://github.com/leviathan747/rosbridge_suite/commit/b0a30dbe002f5ed1262b0b32e476e2d173cc509b)).

I don't understand the design... Why must topics get unregistered at all?

mvollrath, 2020-03-09:
> Why must topics get unregistered at all?

This is a good question, and I don't have a good answer.  Today I'm finding that clients have issues when reconnecting after their topics have been unregistered.  Never unregistering topics fixes it.

This could be a problem in rospy, I'm curious if the ROS2 edition has the same quirk.

closed 2021-09-17 by github-actions[bot]