RobotWebTools/rosbridge_suite #407: invalid syntax in websocket launch

TomShkurti, 2019-05-20:
## Expected Behavior
I am trying to launch a web-socket server from ros2_ament_cmake branch 

## Actual Behavior
I ran the launch command :
`ros2 launch rosbridge_server rosbridge_websocket.launch`

but I got the following error message 
`[ERROR] [launch.LaunchService]: Caught exception in launch (see debug for traceback): invalid syntax (rosbridge_websocket.launch, line 1)
`

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): crystal
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`):
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): ros2_ament_cmake
  - Tornado Version (`python -c 'import tornado; print tornado.version'`):

justinscorringe, 2019-10-24:
I notice this too, unsure why it gives Line 1 error as that part is correct.

justinscorringe, 2019-10-24:
I think we need to make a python launch file?

jnoyola, 2020-01-07:
It seems like maybe ros2 can't actually use XML launch files. I got it working with the python script.
```
> pip install pillow pymongo tornado
> ros2 run rosbridge_server rosbridge_websocket.py
```

closed 2021-09-17 by github-actions[bot]