RobotWebTools/rosbridge_suite #409: StreamClosedError: Stream is closed

Alabate, 2019-06-08:
The client successfully connect to rosbridge websocket, but publish and subscribe don't work. And we've the following error flooding the logs:

```
[ERROR] [1559827744.950238] /rosbridge_websocket: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbridge_server/websocket_handler.py", line 170, in prewrite_message
    yield self.write_message(message, binary)
  File "/usr/lib/python2.7/dist-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/usr/lib/python2.7/dist-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "<string>", line 3, in raise_exc_info
StreamClosedError: Stream is closed
```

Restarting client doen't change anything.

Note: I actually don't know if only publish or subscribe didn't work. The only thing I know is that our "ping" system which is a publish and a subscribe failed silently on the client. We never received any "pong". However the error flooding the logs make me thing that the problem is not only publish or subscribe but the full rosbridge communication.

## Steps to Reproduce the Problem
No idea. The problem happened on our production robot, didn't have time to investigate more. But It happened at least twice, I will keep this issue updated if I can get more in formations.

## Specifications

  - ROS Version: melodic
  - OS Version: bionic
  - Rosbridge Version: `0.11.0`
  - Tornado Version: 4.5.3

Alabate, 2019-06-09:
I managed to reproduce this multiple times by playing with our client (refreshing it, disconnecting it from wifi), but I didn't find any pattern to reproduce it. It seems however that I was wrong and once this bug append, it's not possible to connect to the websocket anymore..

Note: I was testing with #408 applied. While testing this I noticed #410 so maybe it's related, not tried to reproduce it with websocket timeout enabled yet.

jveitchmichaelis, 2019-06-19:
I've had this when I've connected a server to Ros (via roslibpy) and then killed it.

Possible this is payload related - if you try and send a large message and the client disconnects partway through. Odd though, if the socket gets closed you'd think it should be handled with:

```
except WebSocketClosedError:
            rospy.logwarn('WebSocketClosedError: Tried to write to a closed websocket')
```

Can we just duplicate that line to, eg

```
except StreamClosedError:
            rospy.logwarn('StreamClosedError: stream has been closed. Client disconnected?')
raise
```

Alabate, 2019-07-01:
I close this because we didn't meet this problem again after #410.

closed 2019-07-01 by Alabate