RobotWebTools/rosbridge_suite #411: Different protocols? ROS<->ZeroMQ? Does it make sense?

HWiese1980, 2019-06-11:
Hi folks,

I'm kinda new to `rosbridge` in particular, so I'm not sure if I have gotten my head all the way around its core concepts yet, probably not. Are the available protocols/servers all that is needed to bridge between ROS and whatever other protocol you can come up with? 

Thing is, I need a bridge between ROS and ZeroMQ PUB/SUB. Would that require me to implement another protocol or could the existing ones be used to build the bridge? If I had to implement a new rosbridge server like for websocket, where would I start? What do I need?

Thanks

mvollrath, 2019-06-11:
To use the rosbridge library you just need a bridge between the protocol and the client.

For example when the websocket handler gets a new connection, it [instantiates a RosbridgeProtocol for that client and links the outgoing attribute to a callback for messages coming from rosbridge](https://github.com/RobotWebTools/rosbridge_suite/blob/841c8ce271275261e0691e7e90507a26f637b2b6/rosbridge_server/src/rosbridge_server/websocket_handler.py#L96-L97).  It also [passes messages from the ws client to the protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/841c8ce271275261e0691e7e90507a26f637b2b6/rosbridge_server/src/rosbridge_server/websocket_handler.py#L96-L97).  Finally, [when a ws connection is closed the protocol is finished](https://github.com/RobotWebTools/rosbridge_suite/blob/841c8ce271275261e0691e7e90507a26f637b2b6/rosbridge_server/src/rosbridge_server/websocket_handler.py#L148).

HWiese1980, 2019-06-13:
Ah, that sounds interesting. Thank you. I'll have a look into it. Seems at first glance to make things a lot easier than writing a complete new server.

975150313, 2020-09-25:
I also want to know how ROS uses the mqtt communication protocol

HWiese1980, 2020-09-29:
I was experimenting with MQTT in the ROS bridge in my own fork of this repo. Have a look, the code's still there. https://github.com/HWiese1980/rosbridge_suite. You can find parts of it under `rosbridge_server/src/rosbridge_server/paho_mqtt_handler.py`

closed 2021-09-17 by github-actions[bot]