RobotWebTools/rosbridge_suite #415: Unable to load the manifest for package XXX

ch8839, 2019-06-24:
I am running roslaunch rosbridge_server rosbridge_websocket.launch , but it is useless to send any message through the browser, and the browser can't receive the topic of ros.

Error is as follows:
```
[ERROR] [1561364818.360155]: [Client 0] [id: subscribe:/listener :1] subscribe: Unable to load the manifest for package std_msgs. Caused by: line:
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx
unsupported pickle protocol: 4

[ERROR] [1561364913.851892]: [Client 0] [id: advertise:/turtle1/cmd_vel:2] advertise: Unable to load the manifest for package geometry_msgs. Caused by: line:
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx
unsupported pickle protocol: 4
```
I don't know why, I uninstalled and reinstalled and it didn't work.
There is no problem with my js code and the test on other machines is successful

ramezanifar, 2019-07-01:
Could this possibly be a missing
source path_to_catkin_ws/devel/setup.bash 
before 
roslaunch rosbridge_server rosbridge_websocket.launch ?

mvollrath, 2019-08-13:
Maybe related to [this rosdep issue](https://github.com/ros-infrastructure/rosdep/issues/379)

closed 2021-09-17 by github-actions[bot]