RobotWebTools/rosbridge_suite #416: Feature Request: Provide method for retrieving message definitions

gkjohnson, 2019-07-07:
Just like rosbridge provides a method for requesting [ros topic types](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosapi/src/rosapi/proxy.py#L137) it would be great to provide a method for retrieving ros message definitions using something like the [get_msg_text](https://github.com/strawlab/ros_comm/blob/master/tools/rosmsg/src/rosmsg.py#L229) function the `rosmsg` utility.

Specifically I'm interested in using this with Roslibjs to dynamically infer message types and help to automate some message processing on the client.

Thanks!

uahic, 2019-07-25:
For typechecking however (tyepscript) I dont see how this would help during development time. What I do is to just write interfaces that mirror the message types (that helps alot for coding, especially autocompletion) but yeah... alot of manual typing. Hence -> typescript ;)

gkjohnson, 2021-08-18:
#452 closed this issue

closed 2021-08-18 by gkjohnson

jtbandes, 2021-08-18:
For ROS 2, see also #574 and https://github.com/ros2/ros2/issues/1159