RobotWebTools/rosbridge_suite #418: WebSocketClosedError: Tried to write to a closed websocket

uahic, 2019-07-16:
We upgraded yesterday to rosbridge_suite 0.11.2 and on the webserver side we get now an infinite-repeating of the following warning: 

WebSocketClosedError: Tried to write to a closed websocket 

However, our webapp is still running and roslib.js is not complaining on the client side. I just wonder if that error refers to unsubscribed topics or users that disconnected.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): kinetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): xenial
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.2
  - Tornado Version (`python -c 'import tornado; print tornado.version'`): 4.2.1

miguelfrosa, 2019-07-17:
I was about to open an issue about this too. 

I've been looking for what was causing the same problem on my system with the exact same specifications stated before. 

I had to update some docker containers and in the process has to update rosbridge library from APT. 

After that my systems connect (roscore + rosbridge and roslibjs on client side) but I keep receiving the WRN: WebSocketClosedError: Tried to write to a closed websocket on the server side.  

I believe this is related with [https://github.com/RobotWebTools/rosbridge_suite/pull/413](#413) and/or [https://github.com/RobotWebTools/rosbridge_suite/pull/410](url)

MrFrogDeluxe, 2019-07-25:
I have the same problem, client is not complaining at all

vivkr, 2019-07-30:
Facing the same problem as well

jorgenfb, 2019-07-30:
Experiencing the same issue.

closed 2019-08-01 by mvollrath

xarthurx, 2019-08-16:
Experiencing the same issue.
When would this merge be incorporated into a release package?

mvollrath, 2019-08-16:
`apt` is installing 0.11.3 for me today.

xarthurx, 2019-08-16:
Ubuntu 16.04?

xarthurx, 2019-08-16:
I guess you're using Melodic?
I'm on the Kinetic branch.

commit 55b88174 (2019-08-01): Fixes #418: WebSocketClosedError Spam (#423)

* not raising WebSocketClosedError for old tornado versions