RobotWebTools/rosbridge_suite #421: Install problem

ihs-systems, 2019-07-26:
I have : Ubuntu 18.04.2 LTS / ROS melodic (all updated to latest)

When I try to install ros bridge suite with command: 

sudo apt-get install ros-melodic-rosbridge-server

i get this errors:

 ```
Err:1 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-rosbridge-library amd64 0.11.1-1bionic.20190508.025706
  404  Not Found [IP: 64.50.233.100 80]
Err:2 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-rosapi amd64 0.11.1-1bionic.20190508.030429
  404  Not Found [IP: 64.50.233.100 80]
Err:3 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-rosauth amd64 0.1.7-2bionic.20190320.155717
  404  Not Found [IP: 64.50.233.100 80]
Err:4 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-rosbridge-msgs amd64 0.11.1-1bionic.20190508.025715
  404  Not Found [IP: 64.50.233.100 80]
Err:5 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-rosbridge-server amd64 0.11.1-1bionic.20190508.030720
  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-rosbridge-library/ros-melodic-rosbridge-library_0.11.1-1bionic.20190508.025706_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-rosapi/ros-melodic-rosapi_0.11.1-1bionic.20190508.030429_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-rosauth/ros-melodic-rosauth_0.1.7-2bionic.20190320.155717_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-rosbridge-msgs/ros-melodic-rosbridge-msgs_0.11.1-1bionic.20190508.025715_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-rosbridge-server/ros-melodic-rosbridge-server_0.11.1-1bionic.20190508.030720_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

```

I think that this is since version 0.11.2

Vankir, 2019-08-07:
I also have this issue.

mvollrath, 2019-08-13:
Quite possible you just need to `apt update` because it is looking for the wrong version.  I can't reproduce.

closed 2021-09-17 by github-actions[bot]