RobotWebTools/rosbridge_suite #422: rosbridge crash, GeneratorExit in websocket_handler.py

krixkrix, 2019-07-31:
## Actual Behavior

ros bridge server crashes for so far unknown reasons.
The error I am getting is this:

```
[ERROR] [1564563629.104977]: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbridge_server/websocket_handler.py", line 170, in prewrite_message
    yield self.write_message(message, binary)
GeneratorExit
  (source:/rosbridge_websocket)
```
The above exception traceback is repeated multiple times.
I am not familiar with the coroutine stuff in the websocket_handler.py, but maybe someone has a hint based on the above...?

## Steps to Reproduce the Problem

  1. Error does not occur consistently.
  2. So far I have only seen this when multiple web clients are connected to the same rosbridge

## Specifications

  - ROS Version: melodic
  - OS Version: debian stretch-based Balena Intel NUC image
  - Rosbridge Version: 0.11.1
  - Tornado Version: 4.4.3

mvollrath, 2019-08-01:
You might try this in 0.11.2, that method was changed.

[`GeneratorExit`](https://docs.python.org/3/library/exceptions.html#GeneratorExit) means the consumer (Tornado's `IOLoop`) stopped iterating or explicitly closed the generator.  In this case, I don't think there is anything we can do but hope it happened *after* Tornado finished writing the message, because holding the lock was all we really cared about.

Unless somebody else can reproduce this, I'm assuming the issue is specific to Tornado 4.4.

krixkrix, 2019-08-06:
Ok thanks a lot. I will try the combination of 0.11.2 and Tornado 4.5.3.
Or which Tornado version is required/recommended by 0.11.2?

mvollrath, 2019-08-06:
4.5.3 is supported for melodic/bionic.

krixkrix, 2019-08-08:
My problem was resolved by upgrading to 0.11.2 and Tornado 4.5.3.
Thank you.

closed 2019-08-08 by krixkrix