RobotWebTools/rosbridge_suite #425: rosbridge_server hanging randomly under load

benmitch, 2019-08-08:
## Expected Behavior

rosbridge_server does not hang.

## Actual Behavior

rosbridge_server hangs under some load (several connections, quite high rates (50Hz), but relatively little data throughput).

## Steps to Reproduce the Problem

  1. Run rosbridge_server v 0.11.2 on an RPi 3B+
  1. Connect severally to it from a web browser.
  1. Wait.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): kinetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Raspbian GNU/Linux 9 (stretch)
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.2
  - Tornado Version (`python -c 'import tornado; print tornado.version'`): 4.4.3

We are having trouble with rosbridge_server (0.11.2) freezing when under a fair load (number of connections on a somewhat flaky network, rather than throughput). The freeze appears to be random, and occurs after some period, typically 10-30 seconds.

I investigated, and it looks like it's a race condition for _write_lock, similar to that in issue #403, but not solved by the change of #413 (which is present at 0.11.2).

After some testing, and reading the issue reports, I noticed that somebody had suggested that the behaviour of "yield" inside a "with lock" section wasn't certain. So I tried:

    def prewrite_message(self, message, binary):
        # Use a try block because the log decorator doesn't cooperate with @coroutine.
        try:
            with self._write_lock:
                future = self.write_message(message, binary)

            # When closing, self.write_message() return None even if it's an undocument output.
            # Consider it as WebSocketClosedError
            if future is None:
                raise WebSocketClosedError

            yield future

That is, I un-indented after the line "future =", so the lock is released once the call has been made, but before the yield call.

Immediately, the behaviour seems to be fixed, and it has now been running robustly for several minutes without freezing again. I cannot comment at this stage on whether there have been any side effects, or data corruption, but the upstream signals look fine so far.

mvollrath, 2019-08-08:
[Somebody else was having issues with Tornado 4.4.3](https://github.com/RobotWebTools/rosbridge_suite/issues/422), you might try 4.5.3.  Yielding without the lock disconnects websocket backpressure and can lead to OOM on the server if it can't keep up with outgoing messages.  If your application is not very demanding, this is fine.

benmitch, 2019-08-13:
Thanks for your help with this.

This is a production system and raspbian.raspberrypi.org are currently shipping 4.4.3, so I'm not brave enough at this juncture to pursue that test.

For information, here's our latest:

- We've rationalised the numerous signals moving through rosbridge to a single stream; this seems to eliminate the freezing problem, without the patch, but we haven't soak tested it...

- ...because, for my peace of mind, we're going with the patched version of websocket_handler.py described above.

I'll report back if we learn anything further - thanks again.

mvollrath, 2019-08-13:
I hope to solve most Tornado issues by [removing Tornado from rosbridge](https://github.com/RobotWebTools/rosbridge_suite/pull/426).

reinzor, 2019-09-18:
Experiencing same issues, some data is not being published anymore and some data is keeping old value. Will try the https://github.com/RobotWebTools/rosbridge_suite/pull/426

nickvaras, 2020-02-13:
Same random hanging (melodic, bionic, rosbridge 0.11.3, tornado 4.5.3)

reinzor, 2020-02-13:
What is the "hanging" you experience? We sometimes experience that some old messages are received via the websocket. And some messages are just not coming in anymore on the client side.

nickvaras, 2020-02-13:
The most obvious symptom is that messages stop coming thru, but as a sanity check I've been opening the <IP>:<port> on the browser, and when all is well I get the "AutobahnPython 17.10.1 this is a websocket endpoint" splash screen. But when the random hang happens, that screen doesn't load anymore.  I'm among the people who had to do [this](https://github.com/RobotWebTools/rosbridge_suite/issues/138#issuecomment-60700947), could that have any effect?

As a side note, on the roslibjs side, the .on('connected')/.on('error') events never seem to fire, and the handshakeTimeout on the WebSocket constructor does not seem to do anything...

reinzor, 2020-02-14:
The hanging occurred again but when I visit the websocket page, it seems to work fine:

[image]

Some messages of queues or not coming in anymore. Especially the large ones.

nickvaras, 2020-02-20:
A preliminary observation suggests it may be hanging when I first open the ws from within an infrastructure network and then switch to the local area network where rosbridge server lives and try to open it from there.  First step is to rule out that it is not the router that's somehow hanging. We'll see.

Axel13fr, 2020-02-24:
The hanging occured again with version 0.11.4. In my case, when the websocket hangs, I can still rosnode ping it, I can still net cat to its TCP port but nothing else. For example an HTTP GET on it as above doesn't give anything back. 

And when it starts hanging, clients get disconnected, they can't reconnect and the /connected_clients list doesn't get updated.

mvollrath, 2020-02-28:
I'll be working on an automated stress test to reproduce these kinds of issues.

Axel13fr, 2020-03-01:
@mvollrath here are some more details of all the testing we did and their results. I haven't started diving into the code yet.

We managed to reproduce the problem by doing so:
- setting a ping timeout and a ping interval so that non responsive clients get kicked out by the server
- using a web front end with ROS lib js to connect to the websocket server (which subscribes to about 30 topics and 15 services)
- after a successfull connection from the front end, suddenly disconnect it by for example switching off Wifi (after some time, due to the ping timeout, the server will disconnect the client). The front end won't detect that it's not connected anymore due to long websocket timeouts.
- connect Wifi/network back so the front end suddenly receives a "close" socket.
- when the front end reconnects a few secs later, the server registers the topics as before and then HANGS.

Note that this problem happens with 0.11.3 and 0.11.4 so it seems independant from the websocket implementation. We have had other occurences of this hanging but couldn't reproduce them.

Note as well that we made the following stress test: opening about 10 front end instances at the same time so that the rosbridge server load would go 100% CPU on our embedded PC when registering all topics. This didn't cause any issues. But when we do the same thing with just one client connecting as described above, it did. So I'm not sure at all that this is related to a load issue but rather a TCP socket still alive problem...

mvollrath, 2020-03-02:
Ok, I'll see about automating a WebSocket client abort.  The stress test was working but not revealing anything particularly interesting to this hanging problem.

You might try making sure that *all* topic subscriptions have non-zero `queue_length`, when that value is zero (the default) one client can block all the other clients.  Having a queue decouples it.

nickvaras, 2020-03-02:
Thanks for the update @Axel13fr , that's encouraging progress. When the issue is reproduced, does the issue include the failure to display the "splash" screen, for lack of a better word, when visiting the websocket server on the browser? i.e., http://<ip_address>:9090 

> AutobahnPython 17.10.1 I am not Web server, but a WebSocket Endpoint.

Thanks again!!

Axel13fr, 2020-03-02:
@nickvaras yes it does include the failure of the websocket splash screen when the issue is reproduced.

Axel13fr, 2020-03-04:
@mvollrath I tried a queue length of 1 and of 5 for all my subs but it didn't help. Reproducing the issue is easier when under high load.

The issue doesn't happen under kinetic on 16.04 (partially because its tornado version doesn't kick clients after ping timeout).
 
When setting very high timeouts so that suddenly disconnected clients are not kicked out, on reconnection, the websocket is on high load (probably due to TCP buffers full), ROS messages stop coming BUT the websocket thread is still alive (the autobahn splash screen still responds).

nickvaras, 2020-03-04:
>  ROS messages stop coming BUT the websocket thread is still alive (the autobahn splash screen still responds).

I've seen this too. Furthermore, I've seen what appears to be a frozen topic for a few minutes, only to quickly unravel a bunch of buffered messages after a while, i.e., messages that publish at 1Hz stop coming through, and then all of a sudden the all come in quick succession at a much higher frequency.

Axel13fr, 2020-03-04:
The bunch of buffered messages is due to the following: when the rosbridge server doesn't kick out on timeout, it assumes the client is still connected and continues to write on the TCP socket. The TCP socket OS buffer will fill up so that when your front end / client comes back online, it will receive the whole buffered content as fast as the connection allow transfers for it.

But after that, I've seen the websocket hanging in some cases.

mvollrath, 2020-03-05:
I tried to repro this with a couple of different mobile devices but couldn't quite hit the magic.  Then I tried sending SIGSTOP to a Firefox client and got the condition where other clients are hanging.

The problem in rosbridge is that while a `QueueMessageHandler` thread is blocking in the websocket `OutgoingValve`, it also blocks adding new messages to the queue and pushing off the old ones, which then blocks the rospy topic thread.  This means the queue decoupling I mentioned doesn't actually work.

The solution is to not hold the queue `Condition` while pushing data to the `OutgoingValve`.  Please try [this PR](https://github.com/RobotWebTools/rosbridge_suite/pull/464) and see if it fixes things for you.

Axel13fr, 2020-03-05:
Cheers, I'll thouroughly test it next week on all our cases and variants.

nickvaras, 2020-03-06:
I tried the abrupt switch of WiFi network and it does not hang anymore!. :grin:

Axel13fr, 2020-03-09:
So the PR solves the issue when reconnecting a front end while the rosbridge assumed the connection was still alive (no kick out by timeout).

But if a ping timeout parameter is configured and the sudden disconnection exceeds it;resulting in the rosbridge kicking the client, on reconnection, the hanging still occurs.

mvollrath, 2020-03-10:
This could be the issue with unregistration, when I isolated it today I found that under some conditions a reconnecting client can't register the topics it had previously if it was the only client on those topics.  The known hack for this is to never unregister anything, see https://github.com/RobotWebTools/rosbridge_suite/issues/399

You can try [this branch](https://github.com/EndPointCorp/rosbridge_suite/tree/never_unregister_hack) to see if it fixes the issue now described.  Yet to be seen if this is something we can merge.

Axel13fr, 2020-03-10:
We tested the branch to never unregister anything but it still hangs on reconnection after kicking the client due to timeout.

In this case, the websocket server itself seems to hang as GET HTTP requests on its port do not return the usual blue autobahn banner mentionned above by nickvaras.

nickvaras, 2020-03-10:
Yes, I tested the changes of 464 on a never-unregister clone, and had the hanging occur too.

reinzor, 2020-03-13:
Same here, tested using 464. What we are seeing:

- Autobahn screen is responsive
- Some topics are not coming in anymore
- Latched topics hold old value, are not updating

mvollrath, 2020-03-13:
Anybody who is experiencing this issue while using [this branch](https://github.com/RobotWebTools/rosbridge_suite/issues/138#issuecomment-60700947) (which includes #464) please comment with this survey on which features of rosbridge you are using, and maybe we can find correlation.

```
* [ ] Publisher (publishing from websocket to ROS)
* [ ] Latched Publisher
* [ ] Subscriber (from ROS to websocket)
* [ ] Latched Subscriber
* [ ] Services
* [ ] Actionlib
* [ ] tf republisher
* [ ] rosapi
* [ ] png compression
* [ ] cbor encoding
* [ ] websocket compression
* [ ] websocket ping
* [ ] wifi/other wireless networking
```

The application I'm working on does NOT experience the issue:

* [ ] Publisher (publishing from websocket to ROS)
* [ ] Latched Publisher
* [ ] Subscriber (from ROS to websocket)
* [ ] Latched Subscriber
* [ ] Services
* [ ] Actionlib
* [ ] tf republisher
* [ ] rosapi
* [ ] png compression
* [ ] cbor encoding
* [ ] websocket compression
* [ ] websocket ping
* [ ] wifi/other wireless networking

reinzor, 2020-03-16:
For our application:

ROS Melodic:

* [x] Publisher (publishing from websocket to ROS)
* [ ] Latched Publisher
* [x] Subscriber (from ROS to websocket)
* [x] Latched Subscriber
* [x] Services
* [x] Actionlib
* [ ] tf republisher
* [x] rosapi
* [ ] png compression
* [ ] cbor encoding
* [ ] websocket compression
* [ ] websocket ping
* [x] wifi/other wireless networking
* [ ] Python 3 (when running server)

nickvaras, 2020-03-16:
In mine (does experience hang):

My ROS distro:
* [*] Publisher (publishing from websocket to ROS)
* [*] Latched Publisher
* [*] Subscriber (from ROS to websocket)
* [ ] Latched Subscriber
* [*] Services
* [*] Actionlib
* [*] tf republisher
* [ ] rosapi
* [ ] png compression
* [ ] cbor encoding
* [ ] websocket compression
* [ ] websocket ping
* [*] wifi/other wireless networking
* [ ] Python 3 (when running server)

mvollrath, 2020-03-16:
Thank you, I've updated the survey with "authentication" and "bson encoding", please update if you use either of these features.  Also let me know if there are any more features I'm forgetting or don't know about that might be relevant.

Axel13fr, 2020-04-01:
Hi Gents,
Let's make sure rosbridge doesn't get sick :-) 

My ROS distro: Melodic
* [x] Publisher (publishing from websocket to ROS)
* [ ] Latched Publisher
* [x] Subscriber (from ROS to websocket)
* [x] Latched Subscriber
* [x] Services
* [ ] Actionlib
* [ ] tf republisher
* [ ] rosapi
* [ ] png compression
* [ ] bson encoding
* [ ] cbor encoding
* [ ] websocket compression
* [ ] websocket ping
* [ ] authentication
* [x] wifi/other wireless networking
* [ ] Python 3 (when running server)

nick-paul, 2020-04-24:
The issue consistently occurs with our setup. Rosbridge websocket hangs randomly after a couple of connections.

My ROS distro:
* [x] Publisher (publishing from websocket to ROS)
* [x] Latched Publisher
* [x] Subscriber (from ROS to websocket)
* [ ] Latched Subscriber
* [ ] Services
* [ ] Actionlib
* [ ] tf republisher
* [ ] rosapi
* [ ] png compression
* [ ] bson encoding
* [ ] cbor encoding
* [ ] websocket compression
* [ ] websocket ping
* [ ] authentication
* [ ] wifi/other wireless networking
* [ ] Python 3 (when running server)

BrodieBirko, 2020-05-08:
This issue had only happened once or twice 4 months ago. Now it is happening on a regular basis. I know the amount of topics and services each client is connecting to has increased.

My ROS distro: Melodic
* [x] Publisher (publishing from websocket to ROS)
* [ ] Latched Publisher
* [x] Subscriber (from ROS to websocket)
* [x] Latched Subscriber
* [ ] Services
* [ ] Actionlib
* [ ] tf republisher
* [x] rosapi
* [ ] png compression
* [ ] bson encoding
* [ ] cbor encoding
* [ ] websocket compression
* [ ] websocket ping
* [ ] authentication
* [x] wifi/other wireless networking
* [ ] Python 3 (when running server)

mvollrath, 2020-05-08:
I'm noticing that latching is in common for negative reports.  I've looked at the rosbridge_library latching implementation a few times and nothing jumps out as wrong or even risky, but this doesn't mean the somewhat esoteric usage of rospy (adding and immediately removing a connection) isn't breaking things.

ifollowbenoit, 2020-05-11:
Hello,

I'm going to create a setup to reproduce this issue to try to analyse what is happening. It should be ready this week.

**The setup:**

- One robridge websocket server on device A 
- A robridge websocket client on device B
- 2 access point on the same network switching on and off alternatively (device A and B will switch accordingly)

I will try to test with latched publisher/subscriber to see if it makes any difference. 

Also do you know if there is any way to state that the server is hanging (from server perspective), or to monitor its status in order to create recovery behaviors ? Thanks in advance.

mvollrath, 2020-05-11:
Adding debug logging is the typical way to trace this kind of thing in rosbridge, if doing that I would start at the rosbridge_server level (where websocket interactions and backpressure management happen) and work down into the library to isolate.

In the long term having more permanent debug logging and maybe careful addition of (loud) timeouts where the library blocks could help.

Axel13fr, 2020-05-12:
@ifollowbenoit here are the symptoms we have noticed when hanging:
- new websocket clients not being able to connect
- in some cases, autobahn webpage not responding (GET request instead of websocket upgrade on the websocket port)
So there must be something related to server response not being treated anymore. That's  what @mvollrath is saying in terms I do not deeply understand :-)

basecoplan, 2020-05-12:
Hello!
I am investigating this issue too and maybe have found something interesting
We have old 0.9 version and suffer from oom
Upgrade to 0.11.6 for production brought me to this issue and I began my research
I applied https://github.com/RobotWebTools/rosbridge_suite/issues/212 sync websocket write + https://github.com/RobotWebTools/rosbridge_suite/pull/464 queue deadlock fixes to 0.9 version and started to get this unpredictable behavior again

Working thread analysis shows this strange call stack:
```# ThreadID: 140063218221760
File: "/rosbridge_server/rosbridge_websocket", line 221, in <module>
  IOLoop.instance().start()
File: "/usr/lib/python2.7/dist-packages/tornado/ioloop.py", line 832, in start
  self._run_callback(self._callbacks.popleft())
File: "/usr/lib/python2.7/dist-packages/tornado/ioloop.py", line 605, in _run_callback
  ret = callback()
File: "/usr/lib/python2.7/dist-packages/tornado/stack_context.py", line 277, in null_wrapper
  return fn(*args, **kwargs)
File: "/usr/lib/python2.7/dist-packages/tornado/iostream.py", line 562, in wrapper
  return callback(*args)
File: "/usr/lib/python2.7/dist-packages/tornado/stack_context.py", line 316, in wrapped
  ret = fn(*args, **kwargs)
File: "/usr/lib/python2.7/dist-packages/tornado/websocket.py", line 883, in _on_masked_frame_data
  self._on_frame_data(_websocket_mask(self._frame_mask, data))
File: "/usr/lib/python2.7/dist-packages/tornado/websocket.py", line 920, in _on_frame_data
  handled_future = self._handle_message(opcode, data)
File: "/usr/lib/python2.7/dist-packages/tornado/websocket.py", line 946, in _handle_message
  return self._run_callback(self.handler.on_message, decoded)
File: "/usr/lib/python2.7/dist-packages/tornado/websocket.py", line 494, in _run_callback
  result = callback(*args, **kwargs)
File: "/rosbridge_server/websocket_handler.py", line 129, in on_message
  self.protocol.incoming(message)
File: "/rosbridge_library/protocol.py", line 202, in incoming
  self.operations[op](msg)
File: "/rosbridge_library/capabilities/subscribe.py", line 270, in unsubscribe
  self._subscriptions[topic].unregister()
File: "/rosbridge_library/capabilities/subscribe.py", line 82, in unregister
  self.handler.finish()
File: "/rosbridge_library/internal/subscription_modifiers.py", line 154, in finish
  self.join()
File: "/usr/lib/python2.7/threading.py", line 940, in join
  self.__block.wait()
File: "/usr/lib/python2.7/threading.py", line 340, in wait
  waiter.acquire()
```

Adding timeout to a QueueMessageHandler  ```self.join()``` unblocks rosbridge
Could it be a reason or I've something missed?

mvollrath, 2020-05-12:
I think this explains what the problem is.  To summarize, an incoming message is blocked by an outgoing message.  A hypothesized sequence of events:

1. A client connects to the socket.
2. Client subscribes to some topics.  Data starts flowing to the client.
3. The client can't process messages fast enough and the outgoing backpressure lock is closed **by the IOLoop thread**.
4. The client sends a message to explicitly unsubscribe from a topic to which it is the only subscriber.  This immediately invokes `unregister()` on the `Subscription` **from the IOLoop thread** which synchronously calls `finish()` on the `QueueMessageHandler` which is blocked from writing the rest of its queue because the outgoing backpressure lock is closed.  Only **the IOLoop thread** can open the backpressure lock, but that thread is waiting for the lock.

It's very possible that the backpressure lock doesn't even need to be closed for this to deadlock, just more than zero messages waiting in the `QueueMessageHandler`.  A sudden flush of messages could cause the lock to be closed.

This would happen in all backpressure-protected releases.

Possible specific mitigations:

* When `Subscriber.unregister()` is calling `self.handler.finish()`, run it in a different `Thread` to decouple it from the server thread.
* When `Subscriber.unregister()` is calling `self.handler.finish()`, add a flag to `MessageHandler.finish()` to indicate that it should not publish any more messages or block for any reason.  In `QueueMessageHandler` this would clear the queue.

A more general mitigation:

* Queue all `self.protocol.incoming()` calls at the `rosbridge_server` level and run them in a separate `Thread`.  It would eliminate this entire class of deadlock (incoming message blocked by outgoing messages).  Each server implementation would be responsible for doing this.

mvollrath, 2020-05-12:
Please try #496 and see if it helps.

mvollrath, 2020-05-12:
It looks like the deadlock also happens in `MultiSubscriber.subscribe()` when the topic is latched, I'll add a mitigation for that too.

mvollrath, 2020-05-12:
Specific fix for latching subscriber was looking hairy, so I added the general mitigation in that PR.

J-Rojas, 2020-05-13:
I'm running into this regularly when I publish a high frequency /tf topic at 100hz across the bridge. Client disconnects/reconnects are not possible and the entire bridge process appears to be deadlocked and needs to be killed.

mvollrath, 2020-05-13:
I'll roll out 0.11.7 with #496 and we will see what happens.  Quite possible we will find more concurrency glitches in rosbridge_library since we've closed the door on deadlock and opened the door to race conditions.

nickvaras, 2020-05-14:
Not sure if this is mentioned above, might be old news but I'll post it just so people can find it.
After several minutes of the websocket being in the hung state, the plug finally comes off and all thouse buffered messages come rushing in at warp speed, while the following messages are logged:

[ WARN  ] [/rosbridge_websocket]: StreamClosedError: Tried to write to a closed stream
[ WARN  ] [/rosbridge_websocket]: WebSocketClosedError: Tried to write to a closed websocket

Axel13fr, 2020-05-14:
We have tested #496 under high load, it seems to not hang anymore but we've noticed 2 issues so far:
- some clients manage to receive all their subscribed topics while at the same time, some other newly connected clients get only some topics.
- when using this version in combination with ping timeout parameter, when the websocket kicks a client, the following error happens

`
2020-05-14 12:26:44+0000 [RosbridgeWebSocket,0,192.168.7.200] Unhandled Error
 Traceback (most recent call last):
  File "/workspace/install/lib/rosbridge_server/rosbridge_websocket", line 320, in <module>
    reactor.run()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1243, in run
    self.mainLoop()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1255, in mainLoop
    self.doIteration(t)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/epollreactor.py", line 235, in doPoll
    log.callWithLogger(selectable, _drdw, selectable, fd, event)
 --- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 627, in _doReadOrWrite
    self._disconnectSelectable(selectable, why, inRead)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 255, in _disconnectSelectable
    selectable.connectionLost(f)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 289, in connectionLost
    protocol.connectionLost(reason)
  File "/usr/lib/python2.7/dist-packages/autobahn/twisted/websocket.py", line 128, in connectionLost
    self._connectionLost(reason)
  File "/usr/lib/python2.7/dist-packages/autobahn/websocket/protocol.py", line 2467, in _connectionLost
    WebSocketProtocol._connectionLost(self, reason)
  File "/usr/lib/python2.7/dist-packages/autobahn/websocket/protocol.py", line 1098, in _connectionLost
    self._onClose(self.wasClean, self.remoteCloseCode, self.remoteCloseReason)
  File "/usr/lib/python2.7/dist-packages/autobahn/twisted/websocket.py", line 171, in _onClose
    self.onClose(wasClean, code, reason)
  File "/workspace/install/lib/python2.7/dist-packages/rosbridge_server/autobahn_websocket.py", line 248, in onClose
    self.protocol.finish()
  File "/workspace/install/lib/python2.7/dist-packages/rosbridge_library/protocol.py", line 274, in finish
    capability.finish()
  File "/workspace/install/lib/python2.7/dist-packages/rosbridge_library/capabilities/subscribe.py", line 346, in finish
    subscription.unregister()
  File "/workspace/install/lib/python2.7/dist-packages/rosbridge_library/capabilities/subscribe.py", line 88, in unregister
    manager.unsubscribe(self.client_id, self.topic)
  File "/workspace/install/lib/python2.7/dist-packages/rosbridge_library/internal/subscribers.py", line 219, in unsubscribe
    self._subscribers[topic].unsubscribe(client_id)
 exceptions.KeyError: u'/registered_simulators'
`

mvollrath, 2020-05-14:
Thanks @Axel13fr , now working on thread safety for capabilities.

mvollrath, 2020-05-14:
Please try #500 which adds thread safety at capability level.

Axel13fr, 2020-05-18:
We tested it, the hanging happened on that version: clients cannot connect to the socket anymore and an HTTP get on the autobahn socket port doesn't respond.

ifollowbenoit, 2020-05-18:
We tested out with #496 under very high load, and it does not hang anymore, as @Axel13fr stated. Even after a very long time without any connection it still manages to recover

mvollrath, 2020-05-18:
Thanks @Axel13fr I'll add a handler for the ping kick exception instead as a quick fix, since we need to finish shutting down the connection.

mvollrath, 2020-05-19:
#502 should fix the badness caused by the ping kick error, but it will still be logged.

Axel13fr, 2020-05-19:
@mvollrath we tested #502 and we can still cause it to hang. 
It seems to hang in the following case: when disconnecting a client (cleanly or not(cable unplug)), if the bridge is sending a message from ROS to the front end at that moment, it will hang (with the HTTP get on the autobahn websocket port not responding anymore).

When stopping ROS msg transmission to the bridge subs so that nothing is sent to the front, the hanging didn't happen.

mvollrath, 2020-05-19:
I've updated #502 to finish the protocol in the IncomingQueue thread, this should break the incoming/outgoing loop in onClose and unblock the reactor.  There may be exceptions logged during shutdown.

nickvaras, 2020-05-20:
Thank you @mvollrath for keeping the hope alive. Question: Does #502 include #464  ? I tested today and it seems like an improvement regardging the hanging from sudden disconnection (autobahn splash screen remains accessible, services over websocket continue working), but (at least some) topics cease to come thru with multiple clients (for somewhere between 5 to 10 minutes), even though they are still being published (rostopic hz ...etc).

mvollrath, 2020-05-21:
> Question: Does #502 include #464 ?

Yes.

> I seem to be able to replicate the hanging at will by:

Interesting, and it doesn't happen when reconnecting on the same wifi?  Does rosbridge show the usual log messages for removing then adding the client?

mvollrath, 2020-05-21:
Also if you're not already, make sure all your topics have non-zero `queue_length` or else they're all coupled to the `rospy.Subscriber` thread together.

nickvaras, 2020-05-21:
@mvollrath you are the man! This latest version along explicit, non zero queue_length for topic has finally done it. I did try non zero queue_length on a previous server version and still couldn't get it to stay up. Now I can pull the plug on one client and other clients keep receiving their messages the way one would expect. I've spent the grater part of these last 20 hrs looking into this and finally things are looking up. You have my profound, heartfelt gratitude! This really helped!  :thumbsup:

mvollrath, 2020-05-22:
Releasing #502 in 0.11.8 for kinetic and melodic.  Thanks everybody that has helped track this down and test.

@benmitch 
@reinzor 
@nickvaras 
@Axel13fr 
@BrodieBirko 
@ifollowbenoit 
@basecoplan 
@J-Rojas 

Let me know how it goes of course, my application only uses a small set of rosbridge's features so we depend on your reports.

Axel13fr, 2020-05-26:
Congratulations @mvollrath for supporting us all along on this, we have tested is along with the queue length in the frondend and so far no way to reproduce the hanging ! We will continue production test in the coming weeks and keep you posted.

[3 more replies]

closed 2021-09-17 by github-actions[bot]