RobotWebTools/rosbridge_suite #427: subscribe network..

vusdo, 2019-08-13:
nano @ nano: ~ $ rostopic info / topicshow
Type: std_msgs / String

Publishers:
  * / rosbridge_websocket (http: // localhost: 44001 /)

Subscribers:
  * / nano1_sample (http://192.168.1.19:38203/)
  * / rostopic_24744_1565696272026 (http://192.168.1.19:43167/)
  * / rostopic_5169_1565696896016 (http://192.168.1.82:42321/)

-Publish in the browser.
-At 82, echo is normal.
-Echo does not arrive at 19.
-ping or port is normal.
-When you publish from the console, echo arrives at IP 19.

How do I publish in a browser and echo arrives at IP 19?

## Specification
-ROS version: Kinetic
-OS version: bionic
-Rosbridge version: 0.11.2
Tornado Version: 4.2.1

mvollrath, 2019-08-13:
This rostopic output is very helpful, the problem is you need to set the ROS_IP or ROS_HOSTNAME environment variables before running rosbridge_websocket.  Because its address is "localhost", the listener on 19 can not connect to the rosbridge node (it is trying to connect to localhost which is the wrong address).

http://wiki.ros.org/ROS/EnvironmentVariables#ROS_IP.2BAC8-ROS_HOSTNAME

mvollrath, 2019-08-13:
Closing this because it is not a rosbridge issue, see [ROS Answers](https://answers.ros.org/questions/) if you need more ROS help.

closed 2019-08-13 by mvollrath