RobotWebTools/rosbridge_suite #430: Client unregister while trying to publish a CompressedImage using ROSBridge

pramodak, 2019-09-16:
I am trying to publish a ROS sensor_msgs/CompressedImage using ROSBridge. When I put the JPEG image data (captured from Unity3D platform) in 'data' field of the ROS message and publish it, the ROSBridge silently disconnects the client as below. There are no errors. 
[INFO] [1568642603.085737]: Client connected.  1 clients total.
[INFO] [1568642604.647151]: Client disconnected. 0 clients total.

And when I listen to the topic, I see that I receive one ROS event before the client is disconnected. The text size of the event (captured after redirecting rostopic echo output is ~ 1MB)

The compressed image is 640x480 resolution.

Also, when I do not input any JPEG data, the client does not disconnect and the events are published. Does the size of the image data (~ 1MB) an issue here? - is there a workaround or fix for this issue?

## Expected Behavior
The expected behaviour is that the ROSBridge client stays connected and ROS events are received.

## Actual Behavior
The client disconnects and ROSBridge communication is no more possible.

## Steps to Reproduce the Problem

  1. Capture the screenshot using Unity3D platform and compress using JPEG algorithm.
  2. Update the ROS event parameters (header) and publish the JPEG data.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): kinetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): DISTRIB_CODENAME=xenial
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): <version>0.9.0</version>
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): [twisted, version 16.0.0]

mvollrath, 2019-09-16:
That is a very old version of rosbridge, you should try at least 0.11.3 which should be available on kinetic.

pramodak, 2019-09-16:
Thanks for your response - I tried with 0.11.3 ROSbridge and the result is the same.

pramodak, 2019-09-17:
I was able to solve the issue (though I do not know what was causing it). On the WebSocket client side, I was using .NET websocket package in Unity (which is System.Net.WebSockets). I migrated to websocket-sharp and the issue is not observed with 0.11.3 ROSBridge.

closed 2019-09-17 by pramodak