RobotWebTools/rosbridge_suite #434: Auth not working properly since Autobahn websocket server

reinzor, 2019-09-23:
## Expected Behavior

Authentication should work with multiple clients.

## Actual Behavior

When multiple clients set-up an auth connection, at some point the auth service is not called and the client is not authenticated.

## Steps to Reproduce the Problem

  1. Start websocket server with authentication
  1. Open auth connection with client 1
  1. Open auth connection with client 2
  1. Refresh pages of clients a couple of times

## Specifications

  - ROS Version (`MELODIC`):
  - OS Version (`Ubuntu 18.04 LTS`):
  - Rosbridge Version (`3b375b23a4bc36cdc0d79fab8914def77951ded4`):
  - Twisted Version (`[Twisted, version 17.9.0]`):

mvollrath, 2019-09-23:
@reinzor I can't find documentation about how to run the authenticate service, can you point me at some information so I can try to repro this?

reinzor, 2019-09-24:
This is indeed not properly documented (maybe we should add some documentation in the rosbridge aswell).

Rosauth package doc: http://wiki.ros.org/rosauth

then call the `authenticate` method on the `roslibjs` ROS connection instance

reinzor, 2019-09-24:
However, after more testing, it seems to hang only under heavy load so maybe this might be related to https://github.com/RobotWebTools/rosbridge_suite/issues/425

mvollrath, 2020-01-06:
@reinzor is this still an issue?

reinzor, 2020-01-06:
Yes, for now I reduced the bandwidth and it seems to work fine but for high loads, authentication seems to fail for some reason. It seems that something is "hanging".

mvollrath, 2020-02-11:
I tried to repro this but couldn't get it to auth properly under normal conditions.  I'm happy to debug this when there's documentation on how to set it up.

mvollrath, 2020-03-09:
You might find this works better with https://github.com/RobotWebTools/rosbridge_suite/pull/464

reinzor, 2020-03-10:
We will try this out! Thanks

reinzor, 2020-06-12:
Seemed to be coupled to https://github.com/RobotWebTools/rosbridge_suite/issues/425 , no longer deserves a separate issue.

closed 2020-06-12 by reinzor