RobotWebTools/rosbridge_suite #437: Increasing delays when subscriber callback takes too long

HWiese1980, 2019-10-10:
I'm encountering increasing delays on messages that arrive in a callback that runs longer than the topic's period time. See https://answers.ros.org/question/334898/weird-increasing-delay-in-rosbridge-despite-queue_size1-and-buff_size224 for further details. 
I can reproduce this behavior in `roslibpy` and `roslibjs`, and infer from that that it is somehow caused by `rosbridge`. I can also reproduce it in pure `rospy` without the bridge in between when I leave the `rospy.Subscriber` settings regarding `queue_size` and `buff_size` on default. If I set them like `queue_size=1` and `buff_size=2**24` this behavior does not occur.

## Expected Behavior

If `roslibpy.Topic(..., queue_length = 1)` or `queue_length = 0` is set, messages should always arrive on time. If they cannot be processed because the callback is still processing the previous message, they should be dropped. The callback should always process the newest (!) message. No buffering should take place.

## Actual Behavior

It seems like the messages are still buffered/queued somewhere. See my **Update 2** in the linked question for the places that I've found possible buffers/queues so far.

## Steps to Reproduce the Problem

There is a test script in the linked question under **Update 1**  that has proven to reproduce the behavior. 

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): Kinetic Kame
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Ubuntu 16.04 LTS
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.3
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): [Twisted, version 19.7.0]

HWiese1980, 2019-10-10:
I've added **Update 3** over at answers.ros.org which states that I've now tested `rosbridge_server/rosbridge_udp` which shows the same behavior.

mvollrath, 2019-10-10:
This may be the intrinsics of the client libraries and environments at work.

In Chrome (and probably other browsers?) a WebSocket callback will block all the way back to the server until it finishes.  A good pattern to work around this when doing visualizations is to, instead of doing all your drawing in the subscription handler, just store the most recent data and do the drawing for that data in a `requestAnimationFrame` cycle.  This way you never block the socket, you never draw the same thing more than once per frame, and data you aren't fast enough to render is dropped.  Maybe this pattern can help your use case too.

I can't speak for roslibpy or UDP server though.

HWiese1980, 2019-10-14:
That might be the cause. But I'd expect then that there were some options to disable this behavior. Especially because I can disable it with `rospy` without any further ado, other than adding `queue_size=1` and `buff_size=2**24` to the subscriber.

mvollrath, 2019-10-14:
In the case of websockets, if the OS or client have a queue rosbridge can't do anything about it, the only feedback is the client ACKing messages.  In my testing Chrome will queue quite a lot of data, this can be observed by saturating the websocket with a lot of data against a slow handler then terminating rosbridge.  You will see that messages keep getting handled long after the server has disappeared.  You have to handle websocket callbacks responsibly by deferring anything that might block the socket.

Roslib itself can be the cause of slow callbacks if it is spending a lot of time decoding the message.  Browser profiler should show you where the bottleneck is.

mvollrath, 2019-10-14:
More information about the websocket case [in this answer](https://stackoverflow.com/a/44454960/5939768)

HWiese1980, 2019-10-14:
Please bear in mind that I don't have a browser on the receiver side. It's in fact a `roslibpy` client.

mvollrath, 2019-10-14:
roslibpy is still websockets.  If they follow the websocket spec (which I believe Autobahn is known for) it is not allowed to drop messages once they have been received and most likely has a buffer or queue for incoming messages.  Try unblocking the subscribe callback and see if that solves your problem.

HWiese1980, 2019-10-14:
Okay, yeah, if only that was so simple... but that's my problem.

On the other hand I don't understand how rosbridge via UDP has the same problem. Is there some sort of security layer on top of UDP in rosbridge? 

However, I have now implemented a very basic and very early stage MQTT bridge server that appears not to suffer from this issue.

mvollrath, 2019-10-14:
The nature of UDP is that the server doesn't know anything about what the client has received, so there is no way you could have backpressure without extending the protocol.

If MQTT is working for your use case, that is great.  If you don't feel like writing your own bridge you could try any of the existing rosbridge clients and implement a deferred queue to drop messages on the client side, or just use rate throttling.

HWiese1980, 2019-10-14:
> The nature of UDP is that the server doesn't know anything about what the client has received, so there is no way you could have backpressure without extending the protocol.

Exactly. That's why I'm wondering how the problem can appear with UDP too. There has to be some queue on the receiver side then. I'd love some help investigating this. 

> If MQTT is working for your use case, that is great. 

Yes, MQTT does work and I have my custom bridge. For the main purpose that I've been using rosbridge so far, it is unfortunately too late. The project it is used in ends very soon and I can't change the code anymore. But for the future it might be an option to use the MQTT bridge instead or in conjunction with the Websocket bridge. 

> If you don't feel like writing your own bridge you could try any of the existing rosbridge clients and implement a deferred queue to drop messages on the client side, or just use rate throttling.

That would be an option if I had any means to accelerate the receiver function. I already have a minimal function that just writes the latest message into a variable and triggers an event. And everything else is handled in another coroutine. It's still too slow. 

Again, the thing is, I **expect** a setting like `queue_length = 1` or even `queue_length = 0` to do exactly that for me: fetch the latest message and drop everything that comes in during its processing. I expect it to do that because it does so in `rospy`.

mvollrath, 2019-10-14:
The rosbridge protocol and implementation leaves a lot to be desired when it comes to queues, until recently setting queue_length could have no effect whatsoever, and even now there is an undefined transport library queue on the server side and an undefined queue on the client side for probably all of the client libraries.  Sorry that this is throwing a wrench in your application.  Rosbridge is not rospy.

You can solve this problem quickly with a `deque`, e.g.

```python
d = collections.deque(maxlen=1)

def consumer_thread():
    while True:
        try:
            msg = d.pop()
        except IndexError:
            continue
        """Process msg here."""

def sub_callback(msg):
    d.appendleft(msg)  # Will push existing msg out of the deque.
```

mvollrath, 2019-10-15:
Updating the docs to more accurately reflect what should be expected of the queue parameters would be a low-hanging fruit for solving this issue.

huyhoangk50, 2020-04-17:
@HWiese1980 do you have another solution for this problem?

mvollrath, 2020-04-20:
@HWiese1980 #464 might fix this for you.

nickvaras, 2020-05-20:
> This may be the intrinsics of the client libraries and environments at work.
>
> In Chrome (and probably other browsers?) a WebSocket callback will block all the way back to the server until it [...]

What would then be expected to happen if the connection of one client is lost before that callback has finished. Specifically what happens to the _other_ clients still connected. My experience suggests the also stop receiving their messages.

closed 2021-09-17 by github-actions[bot]