RobotWebTools/rosbridge_suite #438: Websocket ping timeout disconnection doesn't work on Kinetic

Axel13fr, 2019-10-22:
When using the rosbridge_websocket on a rather unreliable network link, one can use the optionnal 		 websocket ping parameters such as : 
<param name="websocket_ping_interval" value="2"/>
<param name="websocket_ping_timeout" value="10"/>

## Expected Behavior
When using these parameters, if the network link breaks, the websocket bridge will disconnect the client after a 10sec timeout. This should work whatever the ROS distro.

## Actual Behavior
The behavior is correct on Melodic but on Kinetic, adding these timeout parameters doesn't disconnect the client after timeout. I suspect this to be related to the tornado websocket version (4.5.3 for Ubuntu18.04 and 4.2.1 for Ubuntu 16.04).

## Steps to Reproduce the Problem

  1. Start a **rosbridge_websocket** using the ros:kinetic-ros-base-xenial docker image with the following options on a machine : <param name="websocket_ping_interval" value="2"/>
<param name="websocket_ping_timeout" value="10"/>
  2. Connect to the rosbridge websocket via another machine as client
  3. Disconnect the network between them. On the server side, it doesn't disconnect the client after the 10sec timeout.

## Specifications

  - ROS Version : Kinetic vs Melodic
  - OS Version: Ubuntu 16.04 vs 18.04
  - Rosbridge Version : 0.11.2 vs 0.11.3
  - Twisted Version: Not relevant in this version as tornado is used. Tornado 4.2.1 vs 4.5.3

mvollrath, 2019-10-22:
This should work better when there is a new release with the Autobahn server.

As a bit of trivia, Tornado [does not test the ping timeout feature](https://github.com/tornadoweb/tornado/blob/4677c54cc18bbfbdf0f4dadf11610fab6203fd63/tornado/test/websocket_test.py#L756).

Axel13fr, 2020-02-23:
Indeed Autobahn solves this. Cheers for that change !

closed 2020-02-23 by Axel13fr