RobotWebTools/rosbridge_suite #440: Errors in calling rosapi (ros2)

justinscorringe, 2019-10-24:
## Expected Behavior

Trying a call_service with rosapi/topics on the ros2 distro, returns this error. The caller works with ros1 installations fine. Is the protocol changed for ros2? 

## Actual Behavior

`[ERROR] [rosbridge_websocket]: [Client 2] [id: rosbridge-control] call_service NonexistentFieldException: Message type rosapi/Topics_Request does not have a field _structure_needs_at_least_one_member
`

## Steps to Reproduce the Problem

  1. Launch rosbridge_server websocket and rosapi
  2. {"op":"call_service","service":"/rosapi/topics","id":"rosbridge-control","args":["rosapi/Topics"]}

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): Dashing
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): ubuntu 18.04
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): ros2 1.0.2
  - Tornado Version (`python -c 'import tornado; print tornado.version'`): 4.5.3

justinscorringe, 2019-10-25:
https://github.com/ros2/ros2cli/issues/297

Could be related to this, have merged the fixes into my build but am getting the same error

justinscorringe, 2019-11-25:
Same error persists across crystal/dashing/eloquent. Rosapi service call no longer works on its own in eloquent, 'type' error.

dirk-thomas, 2019-11-25:
The fact that you are seeing a field `_structure_needs_at_least_one_member` indicates that you are not using the latest code on either of the distributions. Are you using the Debian packages or a from-source build? For the former please include the version number of all the involved packages (e.g. `rosapi`, `rosidl_generator_py`, etc.). For the later please make sure that you are having the latest state of the sources (share the exact hashed shown by `vcs export --exact`) and have fully rebuild them.

dirk-thomas, 2019-11-25:
> Same error persists across crystal/dashing/eloquent.

I tried this on Eloquent. Can you please double check that this is still a problem on Eloquent?

dirk-thomas, 2019-11-25:
Dashing doesn't have a patch which has been for Eloquent atm. I proposed a backport in ros2/rosidl_python#93.

justinscorringe, 2019-11-26:
Purged everything and now that error is gone: however in rosapi/proxy.py; 
get_service_info needs to beupdated to get_service_server_info, and all is good :)

dirk-thomas, 2019-11-26:
> get_service_info needs to beupdated to get_service_server_info

@justinscorringe Can you please create a pull request for the described change.

ros2/ros2cli#408 makes sure the same API works for Dashing too.

justinscorringe, 2019-11-26:
Made PR https://github.com/RobotWebTools/rosbridge_suite/pull/447

justinscorringe, 2019-11-26:
I'm still having an erronous time exercising all of the rosapi calls, providing commands functional on past examples of rosbridge protocol; 
```
[ERROR] [rosbridge_websocket]: Exception calling subscribe callback: 'Time' object has no attribute 'secs'

[ERROR] [rosbridge_websocket]: [Client 3] [id: rosbridge-f245961f3674f9c1] call_service NonexistentFieldException: Message type rosapi/MessageDetails_Request does not have a field _type

[ERROR] [rosbridge_websocket]: Exception calling subscribe callback: 'dict_values' object is not callable

```

I may have to sift through these one at a time to update on either my end or rosbridge

justinscorringe, 2019-11-26:
In rospy/object.utils; should this change be made?
`"fieldnames": ["secs", "nsecs"],`
`"fieldnames": ["seconds", "nanoseconds"],`

justinscorringe, 2019-11-26:
Could you please advise correct format of calling rosapi services in terminal, I am having type errors with eloquent change

dirk-thomas, 2019-11-27:
If you could provide exact ready-to-run steps to reproduce the problem I am happy to try them and provide feedback what needs change.

justinscorringe, 2019-11-27:
Fresh eloquent install, fresh ros2 branch clone of rosbridge and PR #447 applied before colcon build, environments sourced, ros2 run rosapi.py and rosbridge_websocket (launch xml seems to throw error)

rosapi/Topics and subscriptions work correctly and send response i want.

Websocket call (taken from wireshark to be sure)

```{"op":"call_service","service":"/rosapi/message_details","id":"rosbridge-926d1d182dba4969","args":["rosbridge_msgs/msg/ConnectedClients"]}```

Return

```{"op": "service_response", "service": "/rosapi/message_details", "values": "Message type rosapi/MessageDetails_Request does not have a field _type", "result": false, "id": "rosbridge-926d1d182dba4969"}```

Call:

```{"op":"call_service","service":"/rosapi/message_details","id":"rosbridge-b3077bd8dabe6d4b","args":["rosbridge_msgs/msg/ConnectedClients"]}```

Response:

```{"op": "service_response", "service": "/rosapi/message_details", "values": "Message type rosapi/MessageDetails_Request does not have a field _type", "result": false, "id": "rosbridge-b3077bd8dabe6d4b"}```

closed 2021-09-17 by github-actions[bot]

ussefdesouky, 2022-03-09:
Hello, I'm trying to build a 2d map on a web page using rosbridge, but this is my error when trying to subscribe on the /map topic.

[rosbridge_websocket-1] [ERROR] [1646804810.213169857] [rosbridge_websocket]: Exception calling subscribe callback: a bytes-like object is required, not 'str'