RobotWebTools/rosbridge_suite #445: Websocket error on connect

arBalasquide, 2019-11-16:
## Expected Behavior
When trying to connect to ros socket, I get the following error

## Actual Behavior
[ERROR] [1573926188.033498]: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python3.7/site-packages/rosbridge_server/websocket_handler.py", line 64, in wrapper
    return f(*args, **kwargs)
  File "/opt/ros/melodic/lib/python3.7/site-packages/rosbridge_server/websocket_handler.py", line 143, in on_message
    self.protocol.incoming(message)
  File "/opt/ros/melodic/lib/python3.7/site-packages/rosbridge_library/protocol.py", line 120, in incoming
    self.buffer = self.buffer + message_string
TypeError: can only concatenate str (not "bytes") to str

## Specifications

  - ros melodic
  - Manjaro 4.19
  - 0.11.3
  - Twisted Version 19.7.0
  - Python tornado 4.5.3

arBalasquide, 2019-11-16:
Using this fix [#366](https://github.com/RobotWebTools/rosbridge_suite/issues/366#issue-377833741) seems to resolve the issue, however, shouldnt this be merged into the repo?

mvollrath, 2020-01-07:
You should try develop branch, it fixes some Python 3 issues.

closed 2021-09-17 by github-actions[bot]