RobotWebTools/rosbridge_suite #453: Unexpected delay streaming pointcloud

Changliu52, 2020-01-06:
Hi, we are using websocket to stream pointcloud. The rosbridge subscribes to the cloud topic which streams at 30 FPS with about 2 mbytes per frame. The network is able to stream 2 to 3 frames per second. We tried to remove the queueing in the subscriber by changing this line https://github.com/RobotWebTools/rosbridge_suite/blob/21868a3732a400e39d39cbd29c9f9b60f2e21a49/rosbridge_library/src/rosbridge_library/internal/subscribers.py#L93 to `self.subscriber = Subscriber(topic, msg_class, self.callback, queue_size=1, buff_size=2**24)`

## Expected Behavior
We are expecting rosbridge server can stream the latest cloud by dropping the most of the frames our websocket subscriber should receive the cloud with minimal communication delay.

## Actual Behavior
It is dropping the frame without accumulating, but our received cloud has about 3 seconds constant delay. As shown in this video https://youtu.be/iKXJQgimRqc.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): kinetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Ubuntu 16.04
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): develop
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):

We have checked our rendering and websocket subscriber side, the buffer is empty. So the delay has to come from the rosbridge server. I wonder if you could suggest us if there could be some other places where the server could buffering the cloud?

Thank you in advance.

Best Regards,
Chang

mvollrath, 2020-01-07:
This is not normal behavior.

Which compression method are you using for the PointCloud2 topic?  You should use `cbor` for good performance on point clouds (and any other bytestream topics).  The JSON-based compression methods (`none` and `png`) are especially bad at transporting this kind of data and may cause slowdowns on either end.

One thing you can try is, while watching the client side, kill the rosbridge server and see how long it takes to stop animating.  If it stops immediately, there must be some buffering on the server side.  If it continues for some seconds, it must be buffering on the client side.

mvollrath, 2020-02-11:
@Changliu52 what did you find?

Changliu52, 2020-02-12:
Thank you for the information, what we are currently doing is to convert it to image and use png compression and reconstruct the pointcloud in the unity side. Now the compression takes quite a long time. I wonder if there is already implementation in rosbridge to do cbor compression? and how do I configure ros side to encode it?

Changliu52, 2020-02-12:
Also Im using rossharp in unity side. Do we need to decode it ourselves?

mvollrath, 2020-02-12:
I don't see any reason to use PNG compression with Unity, that compression option was made specifically for browsers because they can use tricks to native decode it and WebSocket-level deflate compression features didn't exist at the time.  In fact when transferring bytestreams PNG usually just makes things slower compared to no compression at all.

CBOR is ideal for point cloud bytestreams, but AFAIK there is no ROSSharp implementation to decode it.  The [recommended CBOR library](https://github.com/dahomey-technologies/Dahomey.Cbor) has a [succinct example of deserializing the data](https://github.com/dahomey-technologies/Dahomey.Cbor#deserialization).

Then it would need to be routed through ROSSharp.  The [Deserialize method](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/RosSocket.cs#L219-L233) [usually produces a `JObject`](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/RosSocket.cs#L168), but some condition could produce a `CborObject` instead and [pass the values through](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/RosSocket.cs#L178).  What happens after `Receive` might be more interesting.

Alternatively you could load some websocket library, send the [lower-level protocol](https://github.com/biobotus/rosbridge_suite/blob/master/ROSBRIDGE_PROTOCOL.md) messages to rosbridge yourself and handle the CBOR data from the socket directly.  This would be a quick way to get performant point cloud data in Unity without getting your hands dirty in ROSSharp.

Changliu52, 2020-02-27:
Thank you @mvollrath How much compression rate would you suggestion when using CBOR and how much computational load would this introduce? 

> To get rosbridge to send the CBOR use the "cbor" compression option, messages on "cbor" compressed topics will be binary instead of JSON strings.

Is there anyway to switch on the CBOR compression from the launch file in rosbridge?

> Then it would need to be routed through ROSSharp. The Deserialize method usually produces a JObject, but some condition could produce a CborObject instead and pass the values through. What happens after Receive might be more interesting.

How would you suggest changes in rossharp to be able to receive and decode it? Maybe we should suggest to the developer in rossharp to integrate it?

Changliu52, 2020-02-27:
Just posted a feature request in https://github.com/siemens/ros-sharp/issues/289

mvollrath, 2020-02-29:
> Thank you @mvollrath How much compression rate would you suggestion when using CBOR and how much computational load would this introduce?

CBOR isn't really compression, it's a more efficient encoding than JSON and base64 for point cloud data.  It should also incur less computational load than JSON+base64 because it's doing less work.

> > To get rosbridge to send the CBOR use the "cbor" compression option, messages on "cbor" compressed topics will be binary instead of JSON strings.
> 
> Is there anyway to switch on the CBOR compression from the launch file in rosbridge?

No, the client has to request CBOR when subscribing to a topic.  This way you can choose which topics use CBOR, which is important because it is better for some types of data (e.g. point clouds) than others.

rosbridge_server supports per-topic CBOR starting in version 0.10.0

> How would you suggest changes in rossharp to be able to receive and decode it? Maybe we should suggest to the developer in rossharp to integrate it?

Whenever the incoming data is binary rather than text, handle it with a CBOR decoder instead of JSON.

Changliu52, 2020-03-11:
for the Rosbridge, I have tried to stop the server while streaming. The receiver side immediately stopped, despite the long delay. So I'm sure the queuing up problem is in server side.

Changliu52, 2020-03-11:
@mvollrath for the Rosbridge, I have tried to stop the server while streaming. The receiver side immediately stopped, despite the long delay. So I'm sure the queuing up problem is in server side.

closed 2021-09-17 by github-actions[bot]