RobotWebTools/rosbridge_suite #454: Websocket connection from localfile origin fails

Axel13fr, 2020-01-27:
## Expected Behavior
The websocket shall allow a connection from a local file origin (ex: file://mypage.html), provided the user set a parameter.

## Actual Behavior
The autobahn websocket refuses connections from local origins. It requires an extra option to allow it. The parameter is not accessible from ros params.

## Steps to Reproduce the Problem

  1.Load a local  page which connects the websocket.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Ubuntu 18.04
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 1.0.2
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): 19.10.0

###Solution:
Autobahn convert local file origins to a null origin. One can allow local origins to work by allowing Null origin:
@@ -276,6 +276,7 @@ if __name__ == "__main__":
         perMessageCompressionAccept=handle_compression_offers,
         autoPingInterval=ping_interval,
         autoPingTimeout=ping_timeout,
+        allowNullOrigin=True
     )

boilerbots, 2020-03-12:
Yes please. We run a browser locally to display a status and use the local file option so we don't need to maintain an html server on a robot.

boilerbots, 2020-03-12:
Also, should big changes be made to Melodic, it was released 2 years ago and people might expect it to be stable.

Axel13fr, 2020-03-12:
Check https://github.com/RobotWebTools/rosbridge_suite/pull/463
But it doesn't pass tests on kinetic probably because the autobahn version of kinetic doesn't have this null origin parameter.

boilerbots, 2020-03-13:
> Check #463
> But it doesn't pass tests on kinetic probably because the autobahn version of kinetic doesn't have this null origin parameter.

Well, there could be a branch for Kinetic, Melodic and so on. They all don't have to run the exact same code.

boilerbots, 2020-03-14:
This works in Melodic and 18.04: https://github.com/RobotWebTools/rosbridge_suite/pull/469

closed 2020-05-13 by Axel13fr