RobotWebTools/rosbridge_suite #456: client connects and disconnects as soon a starting to use it

mhallak, 2020-02-09:
roslaunch rosbridge_server rosbridge_websocket.launch
Then start to use the socket via roslibjs or via ros-sharp.

## Expected Behavior
Sends, forwards topics as it was working on Ubuntu 16 with ROS Kinetic
Exactly the same setup but in Ubuntu 16.04, ROS Kinetic, python2.7 worked perfectly.

## Actual Behavior
### Log of rosbridge:
`[rospy.internal][INFO] 2020-02-09 18:48:40,800: topic[/rosout] adding connection to [/rosout], count 0
[tornado.access][INFO] 2020-02-09 18:49:48,497: 101 GET / (192.168.100.61) 1.05ms
[rosout][INFO] 2020-02-09 18:49:48,498: Client connected.  1 clients total.
[tornado.application][ERROR] 2020-02-09 18:49:48,500: Uncaught exception GET / (192.168.100.61)
HTTPServerRequest(protocol='http', host='192.168.100.61:9090', method='GET', uri='/', version='HTTP/1.1', remote_ip='192.168.100.61')
Traceback (most recent call last):
  File "/home/sload/anaconda3/lib/python3.7/site-packages/tornado/web.py", line 1699, in _execute
    result = await result
  File "/home/sload/anaconda3/lib/python3.7/site-packages/tornado/websocket.py", line 278, in get
    await self.ws_connection.accept_connection(self)
  File "/home/sload/anaconda3/lib/python3.7/site-packages/tornado/websocket.py", line 881, in accept_connection
    await self._accept_connection(handler)
  File "/home/sload/anaconda3/lib/python3.7/site-packages/tornado/websocket.py", line 964, in _accept_connection
    await self._receive_frame_loop()
  File "/home/sload/anaconda3/lib/python3.7/site-packages/tornado/websocket.py", line 1118, in _receive_frame_loop
    await self._receive_frame()
  File "/home/sload/anaconda3/lib/python3.7/site-packages/tornado/websocket.py", line 1165, in _receive_frame
    if new_len > self.params.max_message_size:
TypeError: '>' not supported between instances of 'int' and 'NoneType'
`
### Networking: sudo strace -ttt -f -e trace=network -p

`[pid 32132] 1581266970.219146 accept4(7,  <unfinished ...>
[pid 32099] 1581266988.494373 accept4(13, {sa_family=AF_INET, sin_port=htons(55594), sin_addr=inet_addr("192.168.100.61")}, [16], SOCK_CLOEXEC) = 18
[pid 32099] 1581266988.494952 accept4(13, 0x7ffd1f65c2e0, [16], SOCK_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)
[pid 32099] 1581266988.495406 recvfrom(18, "GET / HTTP/1.1\r\nHost: 192.168.10"..., 65536, 0, NULL, NULL) = 472
[pid 32099] 1581266988.497069 sendto(18, "HTTP/1.1 101 Switching Protocols"..., 195, 0, NULL, 0) = 195
[pid 32099] 1581266988.497207 setsockopt(18, SOL_TCP, TCP_NODELAY, [1], 4) = 0
[pid 32099] 1581266988.497259 setsockopt(18, SOL_TCP, TCP_NODELAY, [0], 4) = 0
[pid 32099] 1581266988.497770 setsockopt(18, SOL_TCP, TCP_NODELAY, [1], 4) = 0
[pid 32099] 1581266988.498262 sendto(9, "POST /RPC2 HTTP/1.1\r\nHost: local"..., 145, 0, NULL, 0) = 145
[pid 32099] 1581266988.498348 sendto(9, "<?xml version='1.0'?>\n<methodCal"..., 255, 0, NULL, 0) = 255
[pid 32099] 1581266988.498462 recvfrom(9, "HTTP/1.1 200 OK\r\nServer: BaseHTT"..., 8192, 0, NULL, NULL) = 405
[pid 32153] 1581266988.499659 sendto(8, "\275\0\0\0\2\0\0\0,8@^\305\215\303\35\0\0\0\0\2\24\0\0\0/rosbri"..., 193, 0, NULL, 0) = 193
[pid 32099] 1581266988.500098 recvfrom(18, "\201\376\0\255\204\2652z\377\227]\n\246\217\20\t\361\327A\31\366\334P\37\246\231\20\23\340\227\10X"..., 65536, 0, NULL, NULL) = 2612
[pid 32099] 1581266988.582709 recvfrom(18, "\201\376\0016\22s}ciQ\22\0230I_\23g\21\21\na\33_O0\32\31A(Q\r\26"..., 65536, 0, NULL, NULL) = 638
[pid 32099] 1581266988.582779 recvfrom(18, 0x5610ea0932d0, 65536, 0, NULL, NULL) = -1 EAGAIN (Resource temporarily unavailable)
`

## Steps to Reproduce the Problem

  1. launch the rosbridge
  1. connects some client
  1. wonder what's going on

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.3
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): Twisted, version 19.10.0

mhallak, 2020-02-09:
I cloned the repository and went back in time: 0.8.6 and the bridge is working perfectly with ROS melodic, and Ubuntu 18.04 - bionic.
I eventually will try to advance version but if you know the solution of my problem, it would help me!

mhallak, 2020-02-10:
I tried 0.9.0 and this one already is not working for me unfortunately...

mhallak, 2020-02-13:
This issue is still open. I would be thrilled to get some help... some orientation about the problem I have... In addition by mistake I took a new version of ros# and I have the same problem with the both versions of rosbridge_server web socket... So there must be something very basic that escapes me.

mvollrath, 2020-02-13:
A couple of things to try:

* With rosbridge 0.11.3 or older, make sure you're using python-tornado from apt instead of a pip Tornado package.  They change their APIs a lot and we're not supporting all versions of Tornado.
* Try running rosbridge from develop branch.

mhallak, 2020-02-20:
It seems that the problem was related to anaconda environment.
There is a kind of mismatch between the both environments in those versions of rosbridge.
As soon as I removed anaconda/bin from $PATH, everything was fine...

closed 2020-02-20 by mhallak

Sun1307010608, 2020-11-23:
I occured the same problem,  like this "[INFO] [1606110859.784252]: begin to create map subscriber
[INFO] [1606110859.784480]: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
[INFO] [1606110859.784704]: Client connected.  1 clients total.
[INFO] [1606110859.788616]: Client 192.168.0.170 disconnected. 0 clients total.
"
and i reinstall the tornado version from 5.1.1 to 4.1.1 and it works!, hope this will be helpful