RobotWebTools/rosbridge_suite #457: Dual bridges don't allow for filters on one bridge and not the other

matthews-jca, 2020-02-11:
##  Expected Behavior
Two bridges come up, one that launches the standard bridge, one with whitelist filtered topics/services (or below, no topics or services at all)

## Actual Behavior
Two bridges launch as expected
The second API node attempts to launch, but the use of global rosapi calls means the second bridge can hit all topics and services the first bridge can

## Steps to Reproduce the Problem
Run two bridges with the following style launch:

<launch>
  <include file="$(find rosbridge_server)/launch/rosbridge_websocket.launch">
    <arg name="port" value="9090" />
  </include>
  <include ns="readonly" file="$(find rosbridge_server)/launch/rosbridge_websocket.launch">
    <arg name="port" value="9091" />
     <param name="topics_glob" value="" />
     <param name="services_glob" value="" />
     <param name="params_glob" value="" />
  </include>
</launch>

This appears to be a consequence of https://github.com/RobotWebTools/rosbridge_suite/issues/78 where the namespaced api node isn't being hit by the namespaced bridge

## Specifications

  - ROS Version: melodic
  - OS Version: bionic
  - Rosbridge Version: 0.11.3
  - Twisted Version : 17.9.0

reinzor, 2020-02-14:
We are using the exact set-up you described and the topics / services / params are filtered in the `rosbridge` itself. Why is this not working?

matthews-jca, 2020-02-26:
So I found my issue here:
Setting a glob to "" or "[]" or anything else "empty" makes the filter assume allow all instead of allow none. By entering a non-empty glob the filtering can apply to one bridge websocket and not another.
It does mean the rosapi node always returns all the available, so we have to look at the glob  params to get a better sense of what nodes are allowed through a bridge from the client side, but that shouldn't be a big issue.

closed 2020-02-26 by matthews-jca