RobotWebTools/rosbridge_suite #459: Unknown operation: auth. For authentication enabled rosbridge_websocket

naimen, 2020-02-14:
## Expected Behavior
When calling ros.authentication() followed by ros.connect() from roslibjs, the rosbridge should first authenticate the client, then decide to accept the connection or not.

## Actual Behavior
[INFO] [1581685869.479833]: Client connected.  1 clients total.
[INFO] [1581685869.481713]: Awaiting proper authentication...
[WARN] [1581685869.489606]: Client 1 did not authenticate. Closing connection.
[WARN] [1581685869.492439]: Client 1 did not authenticate. Closing connection.
[ERROR] [1581685869.495025]: [Client 1] Unknown operation: auth.  Allowed operations: ['service_response', 'unadvertise_service', 'call_service', 'publish', 'fragment', 'subscribe', 'advertise_service', 'unsubscribe', 'unadvertise', 'advertise']
[INFO] [1581685869.497457]: Client disconnected. 0 clients total

## Steps to Reproduce the Problem
let the secrete be `f1JlG2fNnJIHgjnL` 

A launch file like below, and launch it 
```
<include file="$(find rosbridge_server)/launch/rosbridge_websocket.launch" >
      <arg name="port" value="9000"/>
      <arg name="address" value="0.0.0.0"/>
      <arg name="authenticate" value="true"/>
  </include>

 <node name="rosauth" pkg="rosauth" type="ros_mac_authentication" respawn="true" respawn_delay="30" output="screen">
        <param name="~secret_file_location" value="path-to-secret"/>
    </node>
```
Open the following html page in a browser with roslib.js loaded and get into the console and type
```
var ros = new ROSLIB.Ros();
ros.authenticate("01C0D1F9FB2E945DFE8E19F66072BFC2115518E81D805BCBA6F427D94995181D6725DF71B74C536B1CC43D5133E1CEB81999DA2AA2768CA20363DFB6797B941E","","","random",{secs: 0, nsecs: 0},"",{secs: 0, nsecs: 0})
ros.connect("ws://127.0.0.1:9000")
```

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`):0.10.1
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):[Twisted, version 17.9.0]

matthews-jca, 2020-02-25:
This actually appears to be related to https://github.com/RobotWebTools/rosbridge_suite/issues/134

If the authentication fails for any reason (file path, invalid parameters passed, etc) instead of saying so in the logs the auth operation is passed to the rosbridge protocol.incoming. The op for auth isn't handled as a rosbridge capability, so it then fails that check and displays the error for "unknown operation: auth"

reinzor, 2020-03-16:
I recently updated the `rosauth` wiki with a usage (also in combination with roswebtools): http://wiki.ros.org/rosauth

closed 2021-09-17 by github-actions[bot]

interose, 2022-03-14:
Unfortunately I have the same problem. I followed the steps within the wiki. If I call the auth service via command line, I get a "authenticated: True". But if I try to execute it via roslibjs within the browser, I get the following output by the rosbridge

```
2022-03-14 13:55:14+0000 [-] [INFO] [1647266114.617016]: Client connected.  1 clients total.
2022-03-14 13:55:14+0000 [-] [INFO] [1647266114.621224]: Awaiting proper authentication...
2022-03-14 13:55:14+0000 [-] [ERROR] [1647266114.640886]: [Client 8] Unknown operation: auth.  Allowed operations: ['service_response', 'unadvertise_service', 'call_service', 'publish', 'fragment', 'subscribe', 'advertise_service', 'unsubscribe', 'unadvertise', 'advertise']
2022-03-14 13:55:14+0000 [-] [WARN] [1647266114.649799]: Client 8 did not authenticate. Closing connection.
2022-03-14 13:55:14+0000 [-] [INFO] [1647266114.665145]: Client disconnected. 0 clients total.
```

Here is the corresponding JS

```
const ros = new ROSLIB.Ros();
ros.authenticate("19d9d2166799f1ffd6fee6379f957502aff8716bfebc8cc8b3bac57ade14441bb9678be89d0a7eec9c81291f854d754d7a4de2278bede56f162c2faeb468c68a","client","dest","rand",{secs: 0, nsecs: 0},"level",{secs: 0, nsecs: 0});
ros.connect("ws://127.0.0.1:8080");

ros.on('error', function(error) { console.log( error ); });
ros.on('connection', function() {console.log( "connected" );});

const listener = new ROSLIB.Topic({
    ros : ros,
    name : '/gps_state',
    messageType : 'gps_common/GPSFix'
});
listener.subscribe(function(message) {
    console.log(message.latitude + '  ' + message.longitude);
});
```

I doublechecked the result with disabled authentication, and with that - latitude and longitude values appear in the debug output. With activated authentication still "connected" appears, but the listener will not be called.

Is there something else I could check in order to get it working?