RobotWebTools/rosbridge_suite #465: Websocket connection Failed when switching network

ifollowbenoit, 2020-03-05:
## The current Setup

Hi, i'm using rosbridge websocket with this configuration

  1. Rosbridge server running on a device that often switch wifi it is connected to (multiple access points that are on the same network)
  2. A Rosbridge Client running on another device
  3. Sending punctually rosmessage from the Client to the Server

The issue is that, sometimes, because of the wifi switch, the websocket doesn't respond anymore, It  cannot communicate with the rosbridge anymore and it cannot establish a new websocket connection, even with another client (Error in connection establishment: net::ERR_CONNECTION_REFUSED).

## Expected Behaviour

Just a way to recover thanks to the parameters.

## Specifications

  - ROS Version : Melodic
  - OS Version DISTRIB_CODENAME=bionic
  - Rosbridge Version 0.11.3
  - Twisted Version [Twisted, version 17.9.0]

Axel13fr, 2020-03-06:
Very probably the same issue : #425

ifollowbenoit, 2020-03-09:
Yeah ! Thank you very much, it is very likely the same issue

closed 2020-03-09 by ifollowbenoit