RobotWebTools/rosbridge_suite #466: Migration of rosbridge_suite to ROS2

renehoelbling, 2020-03-06:
Using a custom ROS message which is visualized on a website. The rosbridge works fine with ROS1 and the corresponding custom ROS message. Now I try to migrate the the rosbridge_suite from ROS1 to ROS2.

## Expected Behavior
The rosbridge_websocket should transfer the Message to the Client.

## Actual Behavior

[INFO] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
[INFO] [rosbridge_websocket]: Client connected. 1 clients total.
[INFO] [rosbridge_websocket]: [Client 0] Subscribed to /status
[ERROR] [rosbridge_websocket]: Exception calling subscribe callback: 'array.array' object has no attribute 'get_fields_and_field_types'

## Steps to Reproduce the Problem

  1. colcon build --symlink-install (src/ directory including rosbridge_suite and the custom message)
  2. . install/setup.bash 
  3. ros2 run rosbridge_server rosbridge_websocket

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): eloquent
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): DISTRIB_CODENAME=bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): <version>1.0.2</version>
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): [Twisted, version 19.10.0]

dawonn-haval, 2020-04-20:
@jihoonl, What is the plan for merging changes from the dev branch to the ROS2 branch? Any specific plans for Foxy? Is there a better place for these types of questions?

KOGDev, 2020-07-28:
+1 on this, are there any plans to support this bridge going forward in ROS2? Or has ROS2 efforts switched to the rclnodejs based ros2-node-bridge package? I have patches locally that fix some of these type issues (float and bool objects throw the same error as above), but if there's no plans to support the ROS2 version of this package I will switch to the Node based one.

mvollrath, 2020-07-28:
Speaking for myself, not working actively on ROS2 because I don't have any reason to yet.  Happy to merge ROS2 PRs.  If you don't use features unique to this rosbridge, the Node based one will probably work fine.

amacneil, 2021-08-18:
ROS 2 is now supported (see the `ros2` branch) and deb packages will be published for Foxy and Galactic in the next sync.

Foxglove Studio now also supports connecting to ROS 2 systems over rosbridge (see https://github.com/foxglove/studio/pull/1480).

closed 2021-08-18 by amacneil