RobotWebTools/rosbridge_suite #474: Why do we remove callback when subscribing.

huyhoangk50, 2020-03-31:
I have a problem that it is delayed when I subscribe the map topic from roslib (node js). I tried to change queue_size and queue_length but still it didn't work. 

So I am diving deeply into the code to see why does this happen. I found a piece of code in `rosbridge_library/internal/subscribers.py` and it confused me. Why do we remove callback when calling subcribe function.
```
    def subscribe(self, client_id, callback):
        """ Subscribe the specified client to this subscriber.

        Keyword arguments:
        client_id -- the ID of the client subscribing
        callback  -- this client's callback, that will be called for incoming
        messages

        """
        with self.lock:
            self.subscriptions[client_id] = callback
            # If the topic is latched, add_callback will immediately invoke
            # the given callback.
            self.subscriber.impl.add_callback(self.callback, [callback])
            self.subscriber.impl.remove_callback(self.callback, [callback])
    def unsubscribe(self, client_id):
        """ Unsubscribe the specified client from this subscriber

        Keyword arguments:
        client_id -- the ID of the client to unsubscribe

        """
        with self.lock:
            del self.subscriptions[client_id]

```

I think the code should be
`
    def subscribe(self, client_id, callback):
        """ Subscribe the specified client to this subscriber.

        Keyword arguments:
        client_id -- the ID of the client subscribing
        callback  -- this client's callback, that will be called for incoming
        messages

        """
        with self.lock:
            self.subscriptions[client_id] = callback
            # If the topic is latched, add_callback will immediately invoke
            # the given callback.
            self.subscriber.impl.add_callback(self.callback, [callback])

    def unsubscribe(self, client_id):
        """ Unsubscribe the specified client from this subscriber

        Keyword arguments:
        client_id -- the ID of the client to unsubscribe

        """
        with self.lock:
            self.subscriber.impl.remove_callback(self.callback, [self.subscriptions[client_id]])
            del self.subscriptions[client_id]

`
can any body give me the explanation. thanks.

a-nop, 2020-04-06:
It looks like this exists to make a latched topic publish the latched message to a new client.  It is calling `add_callback` on an existing `rospy.Subscriber` which already has a callback of `self.callback` (instantiated [here](https://github.com/RobotWebTools/rosbridge_suite/blob/dc7fcb282d1326d573abe83579cc7d989ae71739/rosbridge_library/src/rosbridge_library/internal/subscribers.py#L98)).  `self.callback` does the usual handling of messages on the topic, which makes sense because it can serialize the message once and iterate over all the clients to send the message to the transport.

That's my understanding of why this is the way it is, I don't guarantee that this isn't breaking something.

mvollrath, 2020-04-06:
It looks like this exists to make a latched topic publish the latched message to a new client.  It is calling `add_callback` on an existing `rospy.Subscriber` which already has a callback of `self.callback` (instantiated [here](https://github.com/RobotWebTools/rosbridge_suite/blob/dc7fcb282d1326d573abe83579cc7d989ae71739/rosbridge_library/src/rosbridge_library/internal/subscribers.py#L98)).  `self.callback` does the usual handling of messages on the topic, which makes sense because it can serialize the message once and iterate over all the clients to send the message to the transport.

That's my understanding of why this is the way it is, I don't guarantee that this isn't breaking something.

jihoonl, 2020-04-08:
You might want to take a look this issue to know the reason behind the logic.

https://github.com/RobotWebTools/rosbridge_suite/issues/1

closed 2020-04-08 by jihoonl