RobotWebTools/rosbridge_suite #476: Can't deserialize CBOR-encoded messages in C#

berkayalpcakal, 2020-04-01:
Hello,

I am trying to add CBOR decoding support to [ROS#](https://github.com/siemens/ros-sharp), where the CBOR-encoded ROS messages are to be deserialized in Unity3D. 

The issue I have is that I cannot get a proper json string from CBOR-encoded byte array.

## Without CBOR compression
1. Receive regularly encoded (compression: none) byte array from rosbridge_suite ( a dummy PointCloud2 message)
2. No need for CBOR decoding, here is the returned json string from the ASCII decoding of the byte array;
```
"{\"topic\": \"/custom_pc2\", \"msg\": {\"width\": 32, \"is_dense\": true, \"point_step\": 0, \"height\": 16, \"header\": {\"stamp\": {\"secs\": 1585773050, \"nsecs\": 469499111}, \"frame_id\": \"debug\", \"seq\": 4659}, \"data\": \"\", \"fields\": [], \"row_step\": 0, \"is_bigendian\": false}, \"op\": \"publish\"}"
```
3. This is the expected json string since it can be deserialized using [Newtonsoft ](https://www.newtonsoft.com/json)or [.NET](https://docs.microsoft.com/en-us/dotnet/standard/serialization/system-text-json-overview) API without any exceptions.

## With CBOR compression
1. Receive CBOR-encoded (comrpession: cbor) byte array from rosbridge_suite (a dummy PointCloud2 message)
2. Convert byte array to json string via the tools [peteroupc/CBOR](https://github.com/peteroupc/CBOR) or[Dahomey.Cbor](https://github.com/dahomey-technologies/Dahomey.Cbor). Here is the obtained json string;
```
"{\"topic\":\"/custom_pc2\",\"msg\":{\"width\":32,\"is_dense\":true,\"point_step\":0,\"height\":16,\"header\":{\"stamp\":{h'73656373':1585771534,h'6E73656373':462513923},\"frame_id\":\"debug\",\"seq\":3400},\"data\":h'',\"fields\":[],\"row_step\":0,\"is_bigendian\":false},\"op\":\"publish\"}"
```
3. Try deserializing via Newtonsoft or .NET API. Here is the exception;

> {"Invalid JavaScript property identifier character: '. Path 'msg.header.stamp', line 1, position 104."}

* The differences between the two different json strings is at the stamp field.

## Specifications

  - ROS Version (`kinetic`):
  - OS Version (`xenial`):
  - Rosbridge Version (`0.11.4`):
  - Twisted Version (`16.0.0`):

asfg84, 2021-12-03:
I encounter the same issue that 'secs' and 'nsecs' in 'header/stamp' are not correct when using cbor compression:
```
stamp:
   110,115,101,99,115: 99082340
   115,101,99,115: 1638521454
```
Rosbridge Version 0.11.13 in melodic (bionic)

However, it works correctly in Version 0.11.13 in noetic (focal).

mvollrath, 2021-12-03:
This seems like a Python 2 problem.

[Here is where time/duration messages are converted](https://github.com/RobotWebTools/rosbridge_suite/blob/ef0e4172667068b4c582a7277fe0b3071fa06aaf/rosbridge_library/src/rosbridge_library/internal/cbor_conversion.py#L72-L77).  You can see that the "secs" and "nsecs" literals are quoted strings.  In Python 2 these would be non-unicode bytestring literals, which would explain the output being a sequence of ASCII codes.

To fix this I recommend making those literals explicitly unicode when Python 2 is used.

closed 2022-07-03 by github-actions[bot]