RobotWebTools/rosbridge_suite #483: Error in ROS2 Eloquent/Dashing with CompressedImage

repa1030, 2020-04-17:
## Expected Behavior
ROS2 CompressedImage receiving via Rosbridge (Branch ros2)

## Actual Behavior
Error:
`[rosbridge_websocket-1] [ERROR] [rosbridge_websocket]: [Client 0] [id: /test/compressed] publish: sensor_msgs/CompressedImage message requires a sequence<uint8> for field data, but got a <class 'str'>`

## Steps to Reproduce the Problem

  1. Start Rosbridge Server
  2. Start Client with CompressedImage Messages
  

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): Eloquent (but same exception in Dashing)
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Ubuntu 18.04 Bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): Current Ros2 Branch
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): 17.9.0

mvollrath, 2020-04-29:
Can you give more details?  E.g. what is the client doing exactly.

repa1030, 2020-04-30:
The Client is from RosSharp. Everything is working fine with Twist, Pose,... Message. Only CompressedImages is showing this error. The serializer is from Newtonsoft (JSON). I thought this could be caused by decoding (msg.decode()) the incoming messages to append them.

mvollrath, 2020-04-30:
So is the client publishing these messages or is it subscribing?

repa1030, 2020-04-30:
The client is publishing the messages as CompressedImage and the Rosbridge Server is subscribing.

mvollrath, 2020-04-30:
You might try encoding the data field with base64, I know that outgoing messages encode binary blobs as base64 strings but have never looked at how incoming binary blobs are handled.

closed 2020-06-02 by repa1030