RobotWebTools/rosbridge_suite #484: rosbridge websocket  0.11.5 error : No module named twisted.python

farshbafdoustar, 2020-04-20:
## Expected Behavior
rosbridge_websocket running

## Actual Behavior
process[rosbridge_websocket-1]: started with pid [122]
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/rosbridge_server/rosbridge_websocket", line 38, in <module>
    from twisted.python import log
ImportError: No module named twisted.python

## Steps to Reproduce the Problem

  1. installing the version ros-kinetic-rosbridge-suite_0.11.5 on Ubuntu Xenial (inside kinetic-ros-base-xenial)
....
Setting up ros-kinetic-rosbridge-msgs (0.11.5-1xenial-20200413-135822+0000) ...
Setting up ros-kinetic-rosbridge-server (0.11.5-1xenial-20200413-140154+0000) ...
Setting up ros-kinetic-rosbridge-suite (0.11.5-1xenial-20200413-140413+0000) ...
.....

## Specifications

  - ROS Version (`echo $ROS_DISTRO`):kinetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): xenial
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.5
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named twisted

mvollrath, 2020-04-20:
I see, the package depends for rosbridge_server are broken in 0.11.5.  Probably something to do with #460.  Maybe the CI needs to be updated to use the new package format.  @sfalexrog

mvollrath, 2020-04-20:
Seems like this doesn't affect melodic, only kinetic.

farshbafdoustar, 2020-04-20:
Thanks for the reply.
I have checked with rosdep from source files "tag:0.11.5" and it seems the following packages are missing as ros-kinetic-rosbridge-suite package dependency :

- python-twisted
- python-twisted-bin
- python-twisted-core
- python-autobahn
- python-tornado
- python-backports.ssl-match-hostname
- python-imaging 
- python-pil

mvollrath, 2020-04-21:
Workaround for this is installing the packages listed by @farshbafdoustar (thank you for the list)

bknight-i3drobotics, 2020-04-22:
Also requires dependencies:
- python-bson
- python-bson-ext

mvollrath, 2020-04-22:
As of today this is affecting melodic packages too.

moriarty, 2020-04-23:
:+1: also experienced this issue today. does the ros build farm support package format 3 for all previous releases?

Is "ROS_PYTHON_VERSION" always set? or should it be defaulting to python2

mvollrath, 2020-04-23:
If nobody has answers within a week or so I'm reverting the change that caused this problem and making a release for the next sync.

moriarty, 2020-04-24:
I see here:

https://github.com/ros/ros_comm/blob/melodic-devel/clients/rospy/package.xml

It’s setup very similar, and works. @jacobperron or @wjwwod when/where must ROS_PYTHON_VERSION must be to have the  dependencies in the Debs set up properly?
Are the dependencies missing because the ROS_PYTHON_VERSION not set when they ran bloom?

ROS1 only uses python2 out of the box so I would assume that is what the debians would ship with python2 for all ROS1 versions, and to use python3 you need to clone export ROS_PYTHON_VERSION, setup a ROS workspace, run rosdep install (this will get the py3 deps) and then compile everything.

wjwwood, 2020-04-24:
> Are the dependencies missing because the ROS_PYTHON_VERSION not set when they ran bloom?

I don't think that matters, but I didn't implement that feature. https://github.com/ros-infrastructure/bloom/pull/551 might have the info you guys need.

mvollrath, 2020-04-27:
It sounds like maybe we just need to bloom with an up-to-date bloom tool?

msmcconnell, 2020-04-27:
@mvollrath Based on your comment ~4 days ago, this will be fixed in the next kinetic sync. Is that correct?

mvollrath, 2020-04-29:
I've released 0.11.6 for [kinetic](https://github.com/ros/rosdistro/pull/24644) and [melodic](https://github.com/ros/rosdistro/pull/24645), they should appear in the next sync for each distribution.

Confirmed that the previous release was made with bloom 0.8.0 which did not have the patch linked by William.  This version will have Python (2) dependencies.

msmcconnell, 2020-05-05:
The newest kinetic sync occurred over this past weekend. The 0.11.6 version was included and this issue was resolved for me.

boilerbots, 2020-05-14:
Is there a way we can get this pushed out to Melodic? This breaks things that are automated for me.

mvollrath, 2020-05-14:
The last melodic sync was on 4/22.  I bloomed this fix on 4/29.  Follow [ROS Discourse](https://discourse.ros.org/c/release/melodic/38) for distro sync updates.

hadiTab, 2020-06-16:
The issue is still there after the latest sync. I'm testing on melodic with release 0.11.9-1 and it's still missing `bson`.

mvollrath, 2020-06-16:
I'm using melodic 0.11.9 on Ubuntu and it works and I see this:

```
$ dpkg -s ros-melodic-rosbridge-library
Package: ros-melodic-rosbridge-library
Status: install ok installed
[...]
Version: 0.11.9-1bionic.20200530.133201
Depends: python-bson, python-pil, ros-melodic-geometry-msgs, ros-melodic-message-runtime, ros-melodic-roscpp, ros-melodic-rosgraph, ros-melodic-rospy, ros-melodic-rosservice, ros-melodic-rostopic, ros-melodic-std-msgs
[...]
```

What should I do to reproduce?

closed 2021-09-17 by github-actions[bot]

Targetarm-Ankit, 2022-08-08:
Hello , I ma using ROS neotic on my system. Using the following commands(basically installing the pakages that is not found one by one) helped me solving my error
1. pip install twisted
2.  pip install autobah
3.  pip install pymongo (bson didn't worked for me )