RobotWebTools/rosbridge_suite #486: Process dies when launching rosbridge_websocket.launch

cristea, 2020-04-21:
## Expected Behavior
Until now i could run the rosbridge_websocket.launch in the package rosbridge_server without any modification and errors. 

## Actual Behavior
After launching the launch file the process dies:

```console
me@ipc002:~$ roslaunch rosbridge_server rosbridge_websocket.launch 
... logging to /home/me/.ros/log/f7444f98-83b2-11ea-986b-645d863a89e1/roslaunch-ipc002-1591.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ipc002:45789/

SUMMARY
========

PARAMETERS
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_websocket/address: 0.0.0.0
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/bson_only_mode: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: None
 * /rosbridge_websocket/params_glob: [*]
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosbridge_websocket/services_glob: [*]
 * /rosbridge_websocket/topics_glob: [*]
 * /rosbridge_websocket/unregister_timeout: 10
 * /rosbridge_websocket/use_compression: False
 * /rosbridge_websocket/websocket_external_port: 9090
 * /rosbridge_websocket/websocket_ping_interval: 0
 * /rosbridge_websocket/websocket_ping_timeout: 30
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

auto-starting new master
process[master]: started with pid [1603]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f7444f98-83b2-11ea-986b-645d863a89e1
process[rosout-1]: started with pid [1618]
started core service [/rosout]
process[rosbridge_websocket-2]: started with pid [1630]
process[rosapi-3]: started with pid [1638]
2020-04-21 11:32:07+0200 [-] Log opened.
2020-04-21 11:32:08+0200 [-] Traceback (most recent call last):
2020-04-21 11:32:08+0200 [-]   File "/opt/ros/kinetic/lib/rosbridge_server/rosbridge_websocket", line 48, in <module>
2020-04-21 11:32:08+0200 [-]     from rosbridge_server import ClientManager
2020-04-21 11:32:08+0200 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_server/__init__.py", line 2, in <module>
2020-04-21 11:32:08+0200 [-]     from .websocket_handler import RosbridgeWebSocket
2020-04-21 11:32:08+0200 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_server/websocket_handler.py", line 49, in <module>
2020-04-21 11:32:08+0200 [-]     from rosbridge_library.rosbridge_protocol import RosbridgeProtocol
2020-04-21 11:32:08+0200 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/rosbridge_protocol.py", line 34, in <module>
2020-04-21 11:32:08+0200 [-]     from rosbridge_library.protocol import Protocol
2020-04-21 11:32:08+0200 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/protocol.py", line 40, in <module>
2020-04-21 11:32:08+0200 [-]     from rosbridge_library.util import json, bson
2020-04-21 11:32:08+0200 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/util/__init__.py", line 19, in <module>
2020-04-21 11:32:08+0200 [-]     import bson
2020-04-21 11:32:08+0200 [-] ImportError: No module named bson
[rosbridge_websocket-2] process has died [pid 1630, exit code 1, cmd /opt/ros/kinetic/lib/rosbridge_server/rosbridge_websocket __name:=rosbridge_websocket __log:=/home/gessmann/.ros/log/f7444f98-83b2-11ea-986b-645d863a89e1/rosbridge_websocket-2.log].
log file: /home/gessmann/.ros/log/f7444f98-83b2-11ea-986b-645d863a89e1/rosbridge_websocket-2*.log
^C[rosapi-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

## Steps to Reproduce the Problem

  1. Today i made a apt-get upgrade
  2. roslaunch rosbridge_server rosbridge_websocket.launch without any modification in the launch file.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): kinetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): DISTRIB_CODENAME=xenial
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): <version>0.11.5</version>
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): [twisted, version 16.0.0]

cristea, 2020-04-21:
Reinstalling python-bson and python-bson-ext solves the issue.

I think the issue is in the package dependencies?

mvollrath, 2020-04-21:
Duplicate #484 

You can workaround this by installing any missing libraries, e.g. python-bson

closed 2020-04-21 by mvollrath