RobotWebTools/rosbridge_suite #488: Autobahn introduced in #426 breaks jrosbridge

moriarty, 2020-04-23:
## Expected Behavior

Breaking changes to wait until the next version

## Actual Behavior

Something which was working stopped working

## Steps to Reproduce the Problem

Any sample code using jrosbridge should be broken.
Also anything that previously set Origin without a protocol, just host:port in the header.
Possibly other things.

## Specifications

  - ROS Version: Melodic
  - Rosbridge Version: breaking version is 0.11.4

mvollrath, 2020-04-23:
Are there any relevant errors in the logs?

moriarty, 2020-04-24:
I can try to setup a mock example to generate a log.

I think I commented in #426 with a link to the function in Autobahn which is complaining.

With wireshark we found the websocket header was not specifying the protocol in the Origin field, so Autobahn fails to parse the host.

Inserting a small proxy between code that was running and the a rosbridge websocket server >= 11.4 and adding the protocol to the header is making it work.

mvollrath, 2020-04-25:
Sorry that this changed when it did.

jrosbridge has a couple of options for both maintaining spec conformance (RFC6454, RFC6455) and working with rosbridge/Autobahn:

* Omit the Origin field entirely.  Autobahn will treat this as "null" origin, which is explicitly allowed by rosbridge since #469 which is released in 0.11.5.
* Specify "null" as the Origin value, similarly supported since #469.  Since jrosbridge is not a browser it is not required to do this, but it is allowed by the spec.

If you haven't already, try jrosbridge against 0.11.5 which should allow missing or null origin by default.

mvollrath, 2020-04-29:
@moriarty is 0.11.5 working for you?

zheolls, 2020-05-16:
> @moriarty is 0.11.5 working for you?

I upgraded to 0.11.5, but it still error.

moriarty, 2020-05-16:
No 0.11.5 does not work. Which is why I titled the issue 0.11.4 & newer.

The two “fixes” state - omit the origin field, or set it to null. But the field was set, it is not omitted nor null, but it is set without the “ws://“

Adding a proxy and appending the “ws://“ seems to be the only way- I don’t see a configuration only change to get back to the original behaviour.  https://github.com/RobotWebTools/rosbridge_suite/pull/426#issuecomment-618523253

Unfortunately I likely won’t look into this any further- the proxy is a quick work around until the use of jrosbridge and rosbridge is removed.

zheolls, 2020-05-17:
@moriarty  I seem to have solved the problem, just update the dependency of jrosbridge.

```xml
<dependency>
	<groupId>org.glassfish.tyrus</groupId>
	<artifactId>tyrus-client</artifactId>
	<version>1.17</version>
</dependency>

<dependency>
	<groupId>org.glassfish.tyrus</groupId>
	<artifactId>tyrus-container-grizzly-client</artifactId>
	<version>1.17</version>
</dependency>

<dependency>
	<groupId>org.glassfish.tyrus</groupId>
	<artifactId>tyrus-server</artifactId>
	<version>1.17</version>
</dependency>
```

moriarty, 2020-05-17:
@zheolls if that works I’m sure @rctoris would not mind a PR to fix this issue https://github.com/rctoris/jrosbridge/issues/26 he’s usually busy and doesn’t actively maintain/test that repo but he might merge it

closed 2021-09-17 by github-actions[bot]