RobotWebTools/rosbridge_suite #495: SyntaxError on win10 ros

zhangzhaoweii, 2020-05-11:
## Expected Behavior
I follow the steps written on [http://wiki.ros.org/Installation/Windows](url)to install ros-melodic on Windows10.Then I create a workspace ， download this repo in it and use  `catkin_make`to make it.
But when I launch the file "rosbridge_tcp.launch",it says:

`E:\rosbridge_ws>roslaunch E:\rosbridge_ws\src\rosbridge_suite-develop\rosbridge_server\launch\rosbridge_tcp.launch bson_only_mode:=True
... logging to C:\Users\dell\.ros\log\8011f830-9366-11ea-b515-f48e38e9a3c4\roslaunch-DESKTOP-TD0FSNH-20340.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://ros1:1893/

SUMMARY

PARAMETERS
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_tcp/authenticate: False
 * /rosbridge_tcp/bson_only_mode: True
 * /rosbridge_tcp/delay_between_messages: 0
 * /rosbridge_tcp/fragment_timeout: 600
 * /rosbridge_tcp/host:
 * /rosbridge_tcp/incoming_buffer: 65536
 * /rosbridge_tcp/max_message_size: None
 * /rosbridge_tcp/params_glob: [*]
 * /rosbridge_tcp/port: 9090
 * /rosbridge_tcp/retry_startup_delay: 5
 * /rosbridge_tcp/services_glob: [*]
 * /rosbridge_tcp/socket_timeout: 10
 * /rosbridge_tcp/topics_glob: [*]
 * /rosbridge_tcp/unregister_timeout: 10
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_tcp (rosbridge_server/rosbridge_tcp)

auto-starting new master
process[master]: started with pid [12940]
ROS_MASTER_URI=http://192.168.2.11:11311
setting /run_id to 8011f830-9366-11ea-b515-f48e38e9a3c4
process[rosout-1]: started with pid [21044]
started core service [/rosout]
process[rosbridge_tcp-2]: started with pid [16948]
process[rosapi-3]: started with pid [20748]
  File "E:\rosbridge_ws\src\rosbridge_suite-develop\rosbridge_server\scripts\rosbridge_tcp", line 1
    ./rosbridge_tcp.py
    ^
SyntaxError: invalid syntax
[31m[rosbridge_tcp-2] process has died [pid 16948, exit code 1, cmd python E:\rosbridge_ws\src\rosbridge_suite-develop\rosbridge_server\scripts\rosbridge_tcp __name:=rosbridge_tcp __log:=C:\Users\dell\.ros\log\8011f830-9366-11ea-b515-f48e38e9a3c4\rosbridge_tcp-2.log].
log file: C:\Users\dell\.ros\log\8011f830-9366-11ea-b515-f48e38e9a3c4\rosbridge_tcp-2*.log[0m`

And the log file is :
`[roslaunch][INFO] 2020-05-11 17:05:03,479: Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2020-05-11 17:05:03,489: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2020-05-11 17:05:03,490: roslaunch starting with args ['C:\\opt\\ros\\melodic\\x64\\bin\\roslaunch', 'E:\\rosbridge_ws\\src\\rosbridge_suite-develop\\rosbridge_server\\launch\\rosbridge_tcp.launch', 'bson_only_mode:=True']
[roslaunch][INFO] 2020-05-11 17:05:03,490: roslaunch env is {'LIB': 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.20.27508\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.20.27508\\lib\\x64;C:\\Program Files (x86)\\Windows Kits\\NETFXSDK\\4.6.1\\lib\\um\\x64;D:\\Windows Kits\\10\\lib\\10.0.17763.0\\ucrt\\x64;D:\\Windows Kits\\10\\lib\\10.0.17763.0\\um\\x64;D:\\Microsoft Visual Studio\\VC98\\mfc\\lib;D:\\Microsoft Visual Studio\\VC98\\lib', 'WINDOWSSDK_EXECUTABLEPATH_X64': 'C:\\Program Files (x86)\\Microsoft SDKs\\Windows\\v10.0A\\bin\\NETFX 4.6.1 Tools\\x64\\', 'SYSTEMROOT': 'C:\\WINDOWS', 'CHOCOLATEYLASTPATHUPDATE': '132313221052431302', 'VSCMD_VER': '16.0.0', 'KMP_DUPLICATE_LIB_OK': 'TRUE', 'VS160COMNTOOLS': 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\', 'VSCMD_ARG_TGT_ARCH': 'x64', 'WINDOWSLIBPATH': 'D:\\Windows Kits\\10\\UnionMetadata\\10.0.17763.0;D:\\Windows Kits\\10\\References\\10.0.17763.0', 'COMSPEC': 'C:\\WINDOWS\\system32\\cmd.exe', 'PYTHONPATH': 'E:/rosbridge_ws/devel\\lib/site-packages;C:/opt/ros/melodic/x64\\lib/site-packages', 'ROS_ROOT': 'C:/opt/ros/melodic/x64/share/ros', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'HOMEDRIVE': 'C:', 'SYSTEMDRIVE': 'C:', 'WINDOWSSDKBINPATH': 'D:\\Windows Kits\\10\\bin\\', 'OPENNI2_REDIST64': 'F:\\PCL1.9.1\\3rdParty\\OpenNI2\\Redist', 'FRAMEWORKVERSION64': 'v4.0.30319', 'PROCESSOR_LEVEL': '6', 'OS': 'Windows_NT', 'FSHARPINSTALLDIR': 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\FSharp\\', 'WINDOWSSDK_EXECUTABLEPATH_X86': 'C:\\Program Files (x86)\\Microsoft SDKs\\Windows\\v10.0A\\bin\\NETFX 4.6.1 Tools\\', 'WINDOWSSDKVERBINPATH': 'D:\\Windows Kits\\10\\bin\\10.0.17763.0\\', 'UCRTVERSION': '10.0.17763.0', 'ROS_HOSTNAME': 'ros1', 'CMAKE_PREFIX_PATH': 'E:/rosbridge_ws/devel;C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64;C:/opt/vcpkg/installed/x64-windows', 'VCTOOLSREDISTDIR': 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Redist\\MSVC\\14.20.27508\\', 'TEMP': 'C:\\Users\\dell\\AppData\\Local\\Temp', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'ROS_LOG_FILENAME': 'C:\\Users\\dell\\.ros\\log\\8011f830-9366-11ea-b515-f48e38e9a3c4\\roslaunch-DESKTOP-TD0FSNH-20340.log', 'VCIDEINSTALLDIR': 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\', 'HOMEPATH': '\\Users\\dell', 'VISUALSTUDIOVERSION': '16.0', 'WINDOWSSDKDIR': 'D:\\Windows Kits\\10\\', 'FRAMEWORKDIR64': 'C:\\Windows\\Microsoft.NET\\Framework64\\', 'LOGONSERVER': '\\\\DESKTOP-TD0FSNH', 'QT_PLUGIN_PATH': 'C:\\opt\\rosdeps\\x64\\plugins', 'PKG_CONFIG_PATH': 'E:/rosbridge_ws/devel\\lib\\pkgconfig;C:/opt/ros/melodic/x64\\lib\\pkgconfig;C:/opt/rosdeps/x64\\lib\\pkgconfig', 'OPENNI2_INCLUDE64': 'F:\\PCL1.9.1\\3rdParty\\OpenNI2\\Include', 'INCLUDE': 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.20.27508\\ATLMFC\\include;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.20.27508\\include;C:\\Program Files (x86)\\Windows Kits\\NETFXSDK\\4.6.1\\include\\um;D:\\Windows Kits\\10\\include\\10.0.17763.0\\ucrt;D:\\Windows Kits\\10\\include\\10.0.17763.0\\shared;D:\\Windows Kits\\10\\include\\10.0.17763.0\\um;D:\\Windows Kits\\10\\include\\10.0.17763.0\\winrt;D:\\Windows Kits\\10\\include\\10.0.17763.0\\cppwinrt;D:\\Microsoft Visual Studio\\VC98\\atl\\include;D:\\Microsoft Visual Studio\\VC98\\mfc\\include;D:\\Microsoft Visual Studio\\VC98\\include', 'VCPKG_ROOT': 'c:\\opt\\vcpkg', 'PYTHONHOME': 'C:\\opt\\python27amd64\\', 'ROS_MASTER_URI': 'http://192.168.2.11:11311', '__VSCMD_PREINIT_INCLUDE': 'D:\\Microsoft Visual Studio\\VC98\\atl\\include;D:\\Microsoft Visual Studio\\VC98\\mfc\\include;D:\\Microsoft Visual Studio\\VC98\\include', 'FRAMEWORKVERSION': 'v4.0.30319', 'TMP': 'C:\\Users\\dell\\AppData\\Local\\Temp', 'COMPUTERNAME': 'DESKTOP-TD0FSNH', 'WINDOWSSDKLIBVERSION': '10.0.17763.0\\', 'USERDOMAIN': 'DESKTOP-TD0FSNH', '__DOTNET_ADD_64BIT': '1', 'ROS_IP': '192.168.2.11', 'ROS_PACKAGE_PATH': 'E:/rosbridge_ws/src;C:\\opt\\ros\\melodic\\x64\\share', 'LD_LIBRARY_PATH': 'E:/rosbridge_ws/devel\\lib;C:/opt/ros/melodic/x64\\lib;C:/opt/rosdeps/x64\\lib', 'DRIVERDATA': 'C:\\Windows\\System32\\Drivers\\DriverData', 'FRAMEWORK40VERSION': 'v4.0', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'CHOCOLATEYINSTALL': 'c:\\opt\\chocolatey', 'ALLUSERSPROFILE': 'C:\\ProgramData', 'USERDOMAIN_ROAMINGPROFILE': 'DESKTOP-TD0FSNH', 'ROS_PYTHON_VERSION': '2', 'USERNAME': 'dell', 'SOPC_KIT_NIOS2': 'E:\\altera\\13.0\\nios2eds', 'PROMPT': '$P$G', 'MGLS_LICENSE_FILE': 'C:\\modeltech64_10.4\\License.txt', 'ONEDRIVE': 'C:\\Users\\dell\\OneDrive', '__VSCMD_PREINIT_PATH': 'C:\\ProgramData\\Oracle\\Java\\javapath;F:\\Scripts\\;F:\\;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\system32\\wbem;C:\\Program Files (x86)\\Windows Kits\\8.1\\Windows Performance Toolkit\\;C:\\Program Files (x86)\\Microsoft SQL Server\\110\\Tools\\Binn\\;C:\\Program Files\\Microsoft SQL Server\\110\\Tools\\Binn\\;C:\\Program Files\\Microsoft SQL Server\\110\\DTS\\Binn\\;F:\\MATLAB\\1\\runtime\\win64;F:\\MATLAB\\1\\bin;D:\\runtime\\win64;D:\\bin;D:\\polyspace\\bin;F:\\MATLAB\\1\\polyspace\\bin;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\dotnet\\;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files\\Microsoft SQL Server\\130\\Tools\\Binn\\;C:\\Program Files\\Microsoft SQL Server\\Clien [... cut ...]
[roslaunch.pmon][INFO] 2020-05-11 17:12:17,539: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 20024)>
[rospy.core][INFO] 2020-05-11 17:12:17,539: signal_shutdown [atexit]`
## Specifications

  - ROS Version (`echo $ROS_DISTRO`):melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`):Windows10 1909（18363.720）
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`):0.11.6
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):

Thanks in advance！

WendellZ524, 2020-05-13:
Hi, I am also looking at rosbridge on win10 system I get a similar problem. 
NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

ROS_MASTER_URI=http://127.0.0.1:11311
process[rosbridge_websocket-1]: started with pid [1008]
process[rosapi-2]: started with pid [10512]
Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\catkin_ws\src\rosbridge_suite\rosbridge_server\scripts\rosbridge_websocket", line 1, in <module>
    rosbridge_websocket.py
NameError: name 'rosbridge_websocket' is not defined
[31m[rosbridge_websocket-1] process has died [pid 1008, exit code 1, cmd python C:\opt\ros\melodic\x64\catkin_ws\src\rosbridge_suite\rosbridge_server\scripts\rosbridge_websocket __name:=rosbridge_websocket __log:=C:\Users\27443\.ros\log\a32c47ee-94ae-11ea-890c-58a023d4fe3a\rosbridge_websocket-1.log].
log file: C:\Users\27443\.ros\log\a32c47ee-94ae-11ea-890c-58a023d4fe3a\rosbridge_websocket-1*.log[0m
[INFO] [1589328856.188000]: Rosapi started

It seems I cannot import the module rosbridge_websocket.
But I do not know how to correct it. Thanks

mvollrath, 2020-05-13:
This is almost definitely an issue with the shell environment from which you are starting roslaunch.  I would look at tutorials and post questions specifically about how to use a development catkin workspace on Windows.

closed 2021-09-17 by github-actions[bot]