RobotWebTools/rosbridge_suite #498: Roadmap for newer ROS2 versions

felixdivo, 2020-05-13:
Currently, [ROS Index](https://index.ros.org/search/?term=rosbridge_suite) lists this package as only supporting `dashing` on RSO2. What are the plans for supporting newer versions? Can such plans already be found somewhere? Our search was sadly not successful.

We in our case were surprised to find that apt does not know `ros-eloquent-rosbridge-suite`.

felixdivo, 2020-06-17:
Guess we'll switch to https://github.com/RobotWebTools/rclnodejs.

amacneil, 2021-08-10:
ROS 2 version of rosbridge will be published soon. Please follow https://github.com/RobotWebTools/rosbridge_suite/issues/536 for updates (that issue is newer but has more activity).

closed 2021-08-10 by amacneil