RobotWebTools/rosbridge_suite #501: Authenticating a client based on IP address

ramezanifar, 2020-05-15:
Is it possible to authenticate a client by checking its IP address?
For example having a white list or black list of IP addresses, and allowing/disallowing a client to connect to the websocket server.
Thank you

mvollrath, 2020-05-15:
No, this is not currently a feature of rosbridge.  I recommend firewall rules like iptables or ufw for IP address black/whitelisting.

reinzor, 2020-05-28:
This could be realized by using an alternative [`rosauth`](http://wiki.ros.org/rosauth) instance that validates the `client_ip`.

mvollrath, 2020-05-28:
That's true, it looks like it would need a small change to the rosbridge_server implementations to use the detected client ip instead of trusting the client field sent by the client.  rosauth would cover confidentiality and integrity.

I still recommend kernel filtering if you are interested in protecting service availability from trivial (or even accidental) DoS.  If you're concerned enough about security to IP whitelist it's probably what you want imho.

closed 2021-09-17 by github-actions[bot]