RobotWebTools/rosbridge_suite #508: failing WebSocket opening handshake

balinttahi, 2020-06-03:
Hi, I am trying to use our NACHI ROS Driver (FlexGui 2.4) with rosbridge. It was working fine with ROS Kinetic (and also older ones), now we moved to ROS Melodic and we are not able to connect anymore. The error message is the following: 

```
failing WebSocket opening handshake ('missing port in HTTP Host header '192.168.212.141' and server runs on non-standard port 9090 (wss = False)')
2020-06-02 15:12:57+0200 [-] dropping connection to peer tcp4:192.168.212.138:1050 with abort=False: missing port in HTTP Host header '192.168.212.141' and server runs on non-standard port 9090 (wss = False)
```

We are running rosbridge on the default 9090 port, WSS is false. 

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.4
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): 17.9.0

gavanderhoorn, 2020-06-03:
Is this a duplicate of #497 (which is a duplicate of #488)?

balinttahi, 2020-06-03:
> Is this a duplicate of #497 (which is a duplicate of #488)?

We are not using jrosbridge, so in this sense, it is not. The error is the same.

mvollrath, 2020-06-03:
This should work by default starting in 0.11.7.

gavanderhoorn, 2020-06-03:
According to [status_page/ros_melodic_default.html?q=rosbridge](http://repositories.ros.org/status_page/ros_melodic_default.html?q=rosbridge) Melodic currently is at `0.11.8`.

@balinttahi: would you have pending updates?

balinttahi, 2020-06-03:
[image]

Same issue with 0.11.8

balinttahi, 2020-06-03:
> This should work by default starting in 0.11.7.

I tried many versions from github to solve the problem, starting from 0.11.4 but I had this problem in all of them. 

Now I installed it with 
`sudo apt-get install ros-melodic-rosbridge-server`
but I still getting the same issue.

mvollrath, 2020-06-03:
I can try to reproduce this if you can point me at the repo for NACHI ROS Driver or submit code for an isolated example client.

Otherwise, here is a brief code review:

[Here is where the error comes from](https://github.com/crossbario/autobahn-python/blob/v17.9.1/autobahn/websocket/protocol.py#L2574-L2579).  You can see that if `externalPort` is `None` (which is now the default value), it skips this check.

[Here is where rosbridge gets the param (defaulting to `None` if it is not a base10 integer)](https://github.com/RobotWebTools/rosbridge_suite/blob/12adad57c3a2a6120733d6325cfc0a325b336029/rosbridge_server/scripts/rosbridge_websocket.py#L103-L108) and [here is where that value is sent to the factory](https://github.com/RobotWebTools/rosbridge_suite/blob/12adad57c3a2a6120733d6325cfc0a325b336029/rosbridge_server/scripts/rosbridge_websocket.py#L291).

mvollrath, 2020-06-03:
The client you are using may not be correctly implementing the `Host` header according to [RFC2616](https://www.w3.org/Protocols/rfc2616/rfc2616-sec14.html#sec14.23):

> A "host" without any trailing port information implies the default port for the service requested (e.g., "80" for an HTTP URL).

The only valid case for not specifying the port is a proxy listening on port 80.  In this case configuring `external_port` parameter to `80` should fix this.

balinttahi, 2020-06-03:
5 minutes ago, I almost said it is working. 

FlexGui 2.4 (NACHI Driver) can connect now to rosbridge, if the `websocket_external_port` is `80`.
But roslib.js can't, because of the following error:

`2020-06-03 23:08:31+0200 [-] failing WebSocket opening handshake ('port 9090 in HTTP Host header 'localhost:9090' does not match server listening port 80')
2020-06-03 23:08:31+0200 [-] dropping connection to peer tcp4:127.0.0.1:48994 with abort=False: port 9090 in HTTP Host header 'localhost:9090' does not match server listening port 80`

Neither a C# code (using simple websocket client):
`2020-06-03 23:09:26+0200 [-] failing WebSocket opening handshake ('port 9090 in HTTP Host header '192.168.212.141:9090' does not match server listening port 80')
2020-06-03 23:09:26+0200 [-] dropping connection to peer tcp4:192.168.212.120:55825 with abort=False: port 9090 in HTTP Host header '192.168.212.141:9090' does not match server listening port 80`

So this "solution" kills everything else (or I am missing something).  Ofc. the listening port can't be 80, because for that, I need to run rosbridge as root.

balinttahi, 2020-06-03:
If I am using ws://localhost:80 as the rosbridge url in roslibjs, nothing happens.

balinttahi, 2020-06-03:
If I run 2 rosbridge in 2 different namespace (1 in root, 1 in lets say ... "legacy"). The legacy has the external_port = 80, the one in the root is a totally default rosbridge, then everything works fine ... ofc I am eating more resources than I should and this is just a brute force workaround ... but at least working. 

If you have a much better solution, where:
- I don't need to run 2 bridges
- I don't need to update the cpp code
- I need only to modify the settings of the rosbridge_server
I am super open to use that... 

Thanks!

mvollrath, 2020-06-03:
I recommend fixing the request headers.  If you have a solution for rosbridge, your PR is welcome.

PapeCoding, 2020-08-10:
I am getting the exact same error message like @balinttahi.
As suggested I tried out the newest version 0.11.9 in melodic. Also clearing the external port number. Still, not working. I would be very thankful, if there would be another solution than changing the request, even if this would be the correct solution. In my case, I can't easily change the Unreal Engines source code, which apparently doesn't comply with RFC2616 in this case.

PapeCoding, 2020-08-11:
This should actually be fixed in newer versions of autobahn https://github.com/crossbario/autobahn-python/commit/80fe02de9754b0898d01d31f2252afdcf9ea2764 . Unfortunately these require python3 which might break other things.

rubenanapu, 2020-10-21:
I had the same problem and solved it with the **websocket_external_port** arg.

The port **40000** in the host was mapped to **9090** in the container. Below is the full command I used:

```
roslaunch rosbridge_server rosbridge_websocket.launch _port:=9090 websocket_external_port:=40000 --screen

```

gonzalocasas, 2021-05-03:
> Tried it out on ROS Noetic and updated Autobahn manually. Works perfectly.

Works! Thanks!

ussefdesouky, 2021-12-18:
Worked when the websocket_external_port:=80, one week trying to figure it out.
`roslaunch rosbridge_server rosbridge_websocket.launch _port:=9090 websocket_external_port:=80 --screen`

closed 2022-07-17 by github-actions[bot]