RobotWebTools/rosbridge_suite #509: Rosbridge runs but unable to connect from python side

krishna1996, 2020-06-03:
import roslibpy
client = roslibpy.Ros(host='localhost', port=9090) 
client.connect()
print('Is ROS connected?', client.is_connected) 

## Expected Behavior
Is ROS connected? true

## Actual Behavior
Is ROS connected? False

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`):
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`):
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`):
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):

closed 2020-06-03 by mvollrath

krishna1996, 2020-06-03:
Thank you. I agree this was a mistake my side. But I still have the issue. 

  File "/Users/name/opt/anaconda3/lib/python3.7/site-packages/roslibpy/ros.py", line 83, in run
    raise Exception('Failed to connect to ROS')

Exception: Failed to connect to ROS

mvollrath, 2020-06-03:
If this is a rosbridge issue I can't diagnose or reproduce it unless you fill out the form detailing which versions are in use.

krishna1996, 2020-06-03:
I use ros melodic. 
I have followed the steps for Ros Bridge installation from here https://roslibpy.readthedocs.io/en/latest/reference/index.html#ros-setup 
I am pretty new to ROS so please help. 
I have followed all the steps in my ubuntu machine as well and my ubuntu machine shows this 'Rosbridge WebSocket server started at ws://0.0.0.0:9090'.
On my python side I use a Macbook here I have used the codes from the reference api link for Roslibpy. 
Please help. thank you.

mvollrath, 2020-06-03:
If the server is on a different machine from the client, `localhost` is probably not the right host to connect to.  My advice is to get it working with server and client on the same machine, then figure out how to solve whatever networking problem you're having.

krishna1996, 2020-06-03:
So basically I am running my ubuntu on a virtual machine and here I run my Rosbridge which I try to access from python outside virtual machine. I have tried giving the IP address instead of the localhost but still I have the same issue as above.

krishna1996, 2020-06-08:
I still have this issue!!

mvollrath, 2020-06-08:
@krishna1996 this is a place for reporting issues with rosbridge itself, you should look at [ROS Answers](https://answers.ros.org/questions/) for questions about rosbridge specifically and/or [Stack Overflow](https://stackoverflow.com/) for general computing questions.