RobotWebTools/rosbridge_suite #512: Exception calling subscribe callback: 'float' object has no attribute 'get_fields_and_field_types'

owen7900, 2020-06-25:
## Expected Behavior
Sending rmf_fleet_msgs/FleetState messages through the websocket to the front end.

## Actual Behavior
Subscribe is failing giving this error message 
```
---
stamp:
  sec: 1593087409
  nanosec: 113872237
level: 40
name: rosbridge_websocket
msg: 'Exception calling subscribe callback: ''float'' object has no attribute ''get_fields_and_field_types'''
file: /home/owen/server_ws/install/rosbridge_library/lib/python3.6/site-packages/rosbridge_library/internal/subscribers.py
function: callback
line: 192
---
```
This only occurs when the FleetState message has robots in the Array of Robot states. when the Array is empty there is no problem.

## Steps to Reproduce the Problem

  1.Use roslibjs to subscribe to a topic publishing an rmf_fleet_msgs/FleetState message
  2.Send a FleetState message with a non-empty RobotState Array

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): Ros2 Eloquent
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Bionic 
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): ros2 branch built from source
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): 17.9.0

owen7900, 2020-06-25:
In further testing I have found this error also occurs with a std_msgs/msg/Float32 message as well. I will continue adding to this issue as I find more msg types that throw this error.

danidask, 2021-08-19:
Same issue

Subscibe to a topic publishing std_msgs/msg/Int32MultiArray from javascript frontend  through websocket
``` javascript
  let test_topic = new ROSLIB.Topic({
    ros: ros,
    name: '/test',
    messageType: 'std_msgs/msg/Int32MultiArray'
  })
  test_topic.subscribe(function(message) {
    console.log(message.data)
  })
```
ros_bridge log:
`[rosbridge_websocket-1] [ERROR] [1629385982.186968150] [rosbridge_websocket]: Exception calling subscribe callback: 'array.array' object has no attribute 'get_fields_and_field_types'`

Specifications
ROS2 Foxy
Docker image Ubuntu 20.04 Focal
Host Ubuntu 18.04 Bionic
Python 3.8.10
rosbridge_server 1.0.2 (ros2 branch built from source)

jtbandes, 2021-08-19:
I'm not able to reproduce this issue on Galactic. Will try Foxy soon.

jtbandes, 2021-08-19:
Are you using multiple subscribers (multiple clients) at once? I can reproduce some issues, although not the exact error message, if I have one cbor-raw subscriber and one json subscriber (the code in your example):

```
[rosbridge_websocket-1] [ERROR] [1629395850.023210277] [rosbridge_websocket]: Exception calling subscribe callback: Traceback (most recent call last):
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscribers.py", line 189, in callback
[rosbridge_websocket-1]     callback(outgoing)
[rosbridge_websocket-1] TypeError: 'dict_values' object is not callable
[rosbridge_websocket-1] 
[rosbridge_websocket-1] [ERROR] [1629395851.020892139] [rosbridge_websocket]: Exception calling subscribe callback: Traceback (most recent call last):
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscribers.py", line 189, in callback
[rosbridge_websocket-1]     callback(outgoing)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 163, in on_msg
[rosbridge_websocket-1]     self.handler.handle_message(msg)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/internal/subscription_modifiers.py", line 70, in handle_message
[rosbridge_websocket-1]     self.publish(msg)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 152, in _publish
[rosbridge_websocket-1]     self.publish(message, self.fragment_size, self.compression)
[rosbridge_websocket-1]   File "/home/parallels/Desktop/rosbridge_suite/install/rosbridge_library/lib/python3.8/site-packages/rosbridge_library/capabilities/subscribe.py", line 333, in publish
[rosbridge_websocket-1]     outgoing_msg = bytearray(encode_cbor(outgoing_msg))
[rosbridge_websocket-1] ValueError: cannot serialize unknown object: std_msgs.msg.Int32MultiArray(layout=std_msgs.msg.MultiArrayLayout(dim=[], data_offset=0), data=[1, 2])
[rosbridge_websocket-1] 
```

danidask, 2021-08-21:
@jtbandes thanks for taking the time to look at it.
I rebuild it from the last source code, and now is working as expected. Its nice to see it’s in active development (the source I had was only few weeks old)
I checked having several clients at once and still works.

closed 2022-02-18 by jtbandes