RobotWebTools/rosbridge_suite #513: Runtime error on ROS noetic

Changliu52, 2020-06-26:
## Expected Behavior
Normal function when client connected

## Actual Behavior
When a client connected it throws the following error
```
2020-06-26 23:49:59+0100 [-] [INFO] [1593211799.425566]: Client connected.  1 clients total.
2020-06-26 23:49:59+0100 [-] Exception in thread Thread-7:
2020-06-26 23:49:59+0100 [-] Traceback (most recent call last):
2020-06-26 23:49:59+0100 [-]   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
2020-06-26 23:49:59+0100 [-]     self.run()
2020-06-26 23:49:59+0100 [-]   File "/opt/ros/noetic/lib/python3/dist-packages/rosbridge_server/autobahn_websocket.py", line 109, in run
2020-06-26 23:49:59+0100 [-]     self.protocol.incoming(msg)
2020-06-26 23:49:59+0100 [-]   File "/opt/ros/noetic/lib/python3/dist-packages/rosbridge_library/protocol.py", line 120, in incoming
2020-06-26 23:49:59+0100 [-]     self.buffer = self.buffer + message_string
2020-06-26 23:49:59+0100 [-] TypeError: can only concatenate str (not "bytes") to str
```

## Steps to Reproduce the Problem

  1. install ros noetic
  2. sudo apt install ros-noetic-rosbridge-suite
  3. roslaunch rosbridge_server rosbridge_websocket address:=....
  4. when the error happens when a new unity ros# connected it gives the above error

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): noetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): DISTRIB_CODENAME=focal
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): <version>0.11.9</version>
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): [Twisted, version 18.9.0] in python3

mvollrath, 2020-06-27:
This might have something to do with combination of binary and text messages from the client.

Changliu52, 2020-07-05:
Thanks @mvollrath but the same client can communicate with melodic rosbridge without problem.

closed 2021-09-17 by github-actions[bot]