RobotWebTools/rosbridge_suite #515: Exception in deserialization of BSON with ROS Noetic

repa1030, 2020-08-02:
##  Expected Behavior
ROSBRIDGE Server works fine with ROSSharp and BSON serialization such as in ROS Melodic.

## Actual Behavior
Getting Error:
[Client 0] Exception in deserialization of BSON

## Steps to Reproduce the Problem
  1. Use current ROSBRIDGE develop branch
  2. Use current ROSSharp branch
  3. Send any message with BSON serialization

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): Noetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): Focal
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.9
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): 18.9.0

RyodoTanaka, 2021-05-27:
I have the same error, does any one have the solution ... ?

nick-paul, 2021-11-01:
#545 Fixes some major python3/BSON issues. Does using that branch resolve the issue?

repa1030, 2021-11-02:
Yes, fixed :)

closed 2021-11-02 by repa1030