RobotWebTools/rosbridge_suite #519: Auth not implemented in TCP server

PapeCoding, 2020-08-10:
## Expected Behavior
Since an `authenticate` argument is provided in the launch file for the TCP server, one would expect the bridge to perform an authentication.

## Actual Behavior
The authentication is not implemented in the TCP server at all, as far as I can see.

## Steps to Reproduce the Problem

1. Start TCP server with `roslaunch rosbridge_server rosbridge_tcp.launch authenticate:=true`
2. Wonder why no authentication is performed, when a client connects

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 0.11.9
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): [Twisted, version 17.9.0]

It would be very nice if the authentication would be added for the TCP server as well!

PapeCoding, 2021-08-19:
This was fixed, since the auth feature was removed entirely...

closed 2021-08-19 by PapeCoding

jtbandes, 2021-08-19:
FYI auth was removed in the ROS 2 branch, not the ROS 1 branch.