RobotWebTools/rosbridge_suite #527: "Error processing request: field names must be a list or tuple type" when calling /rosapi/get_param_names in Python3

stevegolton, 2020-08-20:
/rosapi/get_param_names throws an exception when rosapi is running under Python 3.

Looks like the difference between the return type of `filter()` in python 2 vs python 3. In python 2 the list is evaluated immediately, but in python 3 filter returns a generator object which needs to be evaluated using the list() method or similar before the list is produced.

## Expected Behavior
Calling `/rosapi/get_param_names` should return the list of param names.

## Actual Behavior
`/rosapi/get_param_names` throws an exception. 

## Steps to Reproduce the Problem

  1. Run rosapi with python3.
  2. Call `/rosapi/get_param_names`
  3. See the following exception in the rosapi output:
```
[ERROR] [1597914827.181108]: Error processing request: field names must be a list or tuple type
['Traceback (most recent call last):\n', '  File "<ws>/devel/lib/python3/dist-packages/rosapi/srv/_GetParamNames.py", line 144, in serialize\n    length = len(self.names)\n', "TypeError: object of type 'filter' has no len()\n", '\nDuring handling of the above exception, another exception occurred:\n\n', 'Traceback (most recent call last):\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 636, in _handle_request\n    transport.send_message(response, self.seq)\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 672, in send_message\n    serialize_message(self.write_buff, seq, msg)\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msg.py", line 152, in serialize_message\n    msg.serialize(b)\n', '  File "<ws>/devel/lib/python3/dist-packages/rosapi/srv/_GetParamNames.py", line 153, in serialize\n    except TypeError as te: self._check_types(ValueError("%s: \'%s\' when writing \'%s\'" % (type(te), str(te), str(locals().get(\'_x\', self)))))\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 383, in _check_types\n    check_type(n, t, getattr(self, n))\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 302, in check_type\n    raise SerializationError(\'field %s must be a list or tuple type\' % field_name)\n', 'genpy.message.SerializationError: field names must be a list or tuple type\n']
```

## Specifications

  - ROS Version: noetic
  - OS Version: focal
  - Rosbridge Version: 0.11.9
  - Twisted Version: Twisted, version 18.9.0

stevegolton, 2020-08-20:
Looks like this can be fixed by wrapping the filter generator in the `list()` call to evaluate it immediately:

```python
# Python rosbridge_suite/rosapi/src/rosapi/params.py:line 113
+ return list(filter(lambda x: any(fnmatch.fnmatch(str(x), glob) for glob in params_glob), rospy.get_param_names()))
- return filter(lambda x: any(fnmatch.fnmatch(str(x), glob) for glob in
```

duwudi, 2020-10-26:
> Looks like this can be fixed by wrapping the filter generator in the `list()` call to evaluate it immediately:
> 
> ```python
> # Python rosbridge_suite/rosapi/src/rosapi/params.py:line 113
> + return list(filter(lambda x: any(fnmatch.fnmatch(str(x), glob) for glob in params_glob), rospy.get_param_names()))
> - return filter(lambda x: any(fnmatch.fnmatch(str(x), glob) for glob in params_glob), rospy.get_param_names())
> ```

THANK YOU! I was trying to get [ROS Control Center](https://github.com/pantor/ros-control-center) working and this was the fix, nice work!

stevegolton, 2020-11-21:
@duwudi Aha, I was also trying to use ROS Control Center when I found this issue.

commit 2c5d6302 (2020-11-24): Evaluate filter() generator before returning in get_param_names() which is required when using python3. Fixes #527. (#540)
closed 2020-11-24 by mvollrath