RobotWebTools/rosbridge_suite #530: FieldTypeMismatchException with "float32" fields

AviKenz, 2020-09-14:
Hello everybody :)

I want to control my ROS vehicle with [AckermannDrive.msg](https://github.com/ros-drivers/ackermann_msgs/blob/master/msg/AckermannDrive.msg) but the rosbridge throws a **FieldTypeMismatchException** when i publish a message.

After investigation i find out that the "float32" fields in AckermannDrive are the problem. they are implicitly converted to "float" by python and the Rosbridge fails to find the matching ROS type.

"float64" fields doesnt have this problem since they are converted to "double" by python. I tested it with [Vector3.msg](http://docs.ros.org/melodic/api/geometry_msgs/html/msg/Vector3.html).

Thanks in advance :)

## Expected Behavior
[AckermannDrive.msg](https://github.com/ros-drivers/ackermann_msgs/blob/master/msg/AckermannDrive.msg) should be published.

## Actual Behavior
**FieldTypeMismatchException** is thrown.

## Steps to Reproduce the Problem

  1. Install [ackermann_msgs](https://index.ros.org/r/ackermann_msgs/github-ros-drivers-ackermann_msgs/#dashing) package
  1. Start the rosbridge_server
  1. publish AckermannDrive.msg and observe the rosbridge_server console

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): dashing
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): 1.0.2
  - Twisted Version (`python -c 'import twisted; print twisted.version'`): 17.9.0

mvollrath, 2020-09-14:
Probably related to https://github.com/RobotWebTools/rosbridge_suite/pull/493

AviKenz, 2020-09-16:
@mvollrath thanks for your PROMPT reaction. 😊🙏

hope you will consider @travipross [comment](https://github.com/RobotWebTools/rosbridge_suite/pull/493#issuecomment-689140456) and merge the [PR](https://github.com/RobotWebTools/rosbridge_suite/pull/493) soon. 😊😊😊

I need it for my Masterthesis. 😃😃😃

Thanks in advance !!! 😊🙏

mvollrath, 2020-09-16:
@AviKenz I don't intend to do that.  I don't have a ROS2 environment to develop or test with, so we depend on others to work on ROS2.  If you need it done you should make a PR.

AviKenz, 2020-09-18:
the PR is created. let me know your wishes :)

AviKenz, 2020-09-20:
PROBLEM FIXED !!! :)
Thanks again. you save me a lot of work :) :)

closed 2020-09-20 by AviKenz