RobotWebTools/rosbridge_suite #536: Foxy Release?

JWhitleyWork, 2020-10-08:
## Expected Behavior
This stack is availaable on the ROS Build Farm for supported versions of ROS2.

## Actual Behavior
A `dashing` release exists but not a `foxy` one.

## Dependencies that Need Releasing
- https://github.com/GT-RAIL/rosauth/

JWhitleyWork, 2020-10-08:
I have verified that the stack builds with the source version of `rosauth` on the `ros2` branch.

JWhitleyWork, 2020-12-18:
`rosauth` now has [a release](https://github.com/ros/rosdistro/pull/27767) [pending sync](http://repo.ros2.org/status_page/ros_foxy_default.html?q=SYNC). Can the maintainer of this repo (@mvollrath ?) please consider a ROS Foxy release from the `ros2` branch?

JWhitleyWork, 2021-02-04:
Ping to @mvollrath for Foxy release. `rosauth` is now available in the repositories.

mvollrath, 2021-02-04:
Would like to see a new distro appear in CI tests before releasing the package.

JWhitleyWork, 2021-02-04:
@mvollrath The `ros2` branch already has Foxy jobs, but only on PRs. See https://github.com/RobotWebTools/rosbridge_suite/blob/ros2/.github/workflows/ros_ci.yml.

sageig, 2021-02-18:
Just wanted to note that the Foxy job will build successfully if the setup-ros and action-ros-ci versions are updated from 0.0.25 and 0.0.19, respectively, to 0.1.0. I tested it in [this PR](https://github.com/RobotWebTools/rosbridge_suite/pull/557). However, the dashing and eloquent jobs are failing in the same manner as the kinetic and melodic jobs.

jtbandes, 2021-07-16:
Hi folks, what was the status of this? What will it take in order to get rosbridge-suite released for Galactic/Humble?

amacneil, 2021-07-29:
This was discussed in the ros web tools working group today, and will happen soon.

jtbandes, 2021-08-13:
- galactic: https://github.com/ros/rosdistro/pull/30479
- foxy: https://github.com/ros/rosdistro/pull/30480

closed 2021-08-13 by JWhitleyWork

amacneil, 2021-08-13:
FYI these packages will probably take 1-2 weeks before they are published to apt repos (when the next sync happens). We will also publish a new version of [Foxglove Studio](https://foxglove.dev/) next week with support for ROS 2 rosbridge connections.

JWhitleyWork, 2021-08-13:
Yup, totally fine.