RobotWebTools/rosbridge_suite #544: Rosbridge import error: No Module name builtins

MishkaMN, 2020-12-09:
## Expected Behavior

## Actual Behavior

`carma@ubuntu:/$ roslaunch rosbridge_server rosbridge_websocket.launch port:=9090
... logging to /home/carma/.ros/log/84d6962a-3a53-11eb-9afd-000c29e8fc0e/roslaunch-ubuntu-393.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://ubuntu:45311/
SUMMARY
========
PARAMETERS
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_websocket/address: 0.0.0.0
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/bson_only_mode: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: None
 * /rosbridge_websocket/params_glob: [*]
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosbridge_websocket/services_glob: [*]
 * /rosbridge_websocket/topics_glob: [*]
 * /rosbridge_websocket/unregister_timeout: 10
 * /rosbridge_websocket/use_compression: False
 * /rosbridge_websocket/websocket_external_port: None
 * /rosbridge_websocket/websocket_ping_interval: 0
 * /rosbridge_websocket/websocket_ping_timeout: 30
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)
auto-starting new master
process[master]: started with pid [403]
ROS_MASTER_URI=http://localhost:11311
setting /run_id to 84d6962a-3a53-11eb-9afd-000c29e8fc0e
process[rosout-1]: started with pid [416]
started core service [/rosout]
process[rosbridge_websocket-2]: started with pid [423]
process[rosapi-3]: started with pid [434]
2020-12-09 19:19:57+0000 [-] Log opened.
2020-12-09 19:19:57+0000 [-] Traceback (most recent call last):
2020-12-09 19:19:57+0000 [-]   File "/opt/ros/kinetic/lib/rosbridge_server/rosbridge_websocket", line 48, in <module>
2020-12-09 19:19:57+0000 [-]     from rosbridge_server import ClientManager
2020-12-09 19:19:57+0000 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_server/__init__.py", line 2, in <module>
2020-12-09 19:19:57+0000 [-]     from .websocket_handler import RosbridgeWebSocket
2020-12-09 19:19:57+0000 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_server/websocket_handler.py", line 49, in <module>
2020-12-09 19:19:57+0000 [-]     from rosbridge_library.rosbridge_protocol import RosbridgeProtocol
2020-12-09 19:19:57+0000 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/rosbridge_protocol.py", line 34, in <module>
2020-12-09 19:19:57+0000 [-]     from rosbridge_library.protocol import Protocol
2020-12-09 19:19:57+0000 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/protocol.py", line 34, in <module>
2020-12-09 19:19:57+0000 [-]     from builtins import str
2020-12-09 19:19:57+0000 [-] ImportError: No module named builtins
[rosbridge_websocket-2] process has died [pid 423, exit code 1, cmd /opt/ros/kinetic/lib/rosbridge_server/rosbridge_websocket __name:=rosbridge_websocket __log:=/home/carma/.ros/log/84d6962a-3a53-11eb-9afd-000c29e8fc0e/rosbridge_websocket-2.log].
log file: /home/carma/.ros/log/84d6962a-3a53-11eb-9afd-000c29e8fc0e/rosbridge_websocket-2*.log
`

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - ROS Version (`echo $ROS_DISTRO`):
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`):
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`):
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):

msmcconnell, 2020-12-10:
The fix for this is to install the futures package for python2
```
pip install future
```
I'm not sure if this is something missing from the rosbridge apt binaries or if its expected to be downloaded separately.

mvollrath, 2020-12-11:
This is fixed in 0.11.13

alexytsu, 2020-12-14:
I assume there is going to be a couple weeks wait before this fix will land and be released through the ros debian mirrors? 

Is the only way to patch this in the meantime to uninstall the `apt` version and build from source?

mvollrath, 2020-12-14:
You can install the `future` Python package for Python 2 as a workaround, for most users I would highly recommend installing it via your distro's package manager instead of pip.  e.g. `apt install python-future` on Debian/Ubuntu.

closed 2021-09-17 by github-actions[bot]