RobotWebTools/rosbridge_suite #546: Comparison to simple TCP communication

MartinBischoff, 2020-12-15:
In the [ROS-TCP-Endpoint Repo](https://github.com/Unity-Technologies/ROS-TCP-Endpoint) Unity Technolgoies published a new way to communicate between ROS and external applications that is based on a direct TCP communication and message (de-)serialization and topic publication/subscription via [rospy](http://wiki.ros.org/rospy).
 [Here](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/faq.md) is a performance comparison the TCP communication to rosbridge_suite -based communication.
Is the performance of rosbridge_suite really so much lower? Why? What are the advantages of rosbridge_suite of the simple direct communication? Can you give a suggestion when to use which?

Thank you for providing this great ROS package. The rosbridge_suite ROS API was a great help for our developments in the past years.

mvollrath, 2020-12-15:
> Is the performance of rosbridge_suite really so much lower?

It looks like it is comparing rosbridge base64-encoded JSON payloads to sending raw ROS messages as a means of encoding binary blob data.  Binary encoding is obviously better for this mostly due to the overhead of base64 encoding.  Rosbridge does support CBOR encoding and the new "cbor-raw" technique which sends the raw ROS message just like the other server.  Basically, they compared it to the wrong rosbridge compression feature for that type of data.

Using rosbridge "cbor" or "cbor-raw" compression for binary blobs like images is likely to give you orders of magnitude better end-to-end performance compared to "none" or especially "png".  Because base64 encoding also increases the size of the data by at least 33% it is likely to use a lot less bandwidth too.  However afaik CBOR features are not yet implemented for Unity rosbridge clients, so unless you implement that yourself (or sponsor the work) the other server is likely to work better for transmitting images or other binary data like pointclouds to Unity.

MartinBischoff, 2020-12-16:
Thank you @mvollrath . The ROS# rosbridge client supports BSON compression already.

mvollrath, 2020-12-16:
I've never gotten decent performance out of BSON either on the client or server side.  This may have something to do with a lack of native extensions on the Python side.  There may be room for improvement in the rosbridge implementation.  Is it performing well for you?

Otherwise BSON is not designed to be used as a wire protocol, it's an internal data format of MongoDB and when you look at the details this is apparent.  I don't recommend using BSON with rosbridge for any reason other than it already being implemented.

MartinBischoff, 2020-12-17:
We [integrated BSON support](https://github.com/siemens/ros-sharp/pull/198) triggered by @akirayou 's  pull request when `bson_only_mode` became available for rosbridge_suite. Performance is slightly better than JSON for image data but there is room for improvement.
Is there a CBOR .NET library that you can recommend? I'd go for [this](https://docs.microsoft.com/en-us/dotnet/api/system.formats.cbor?view=dotnet-plat-ext-5.0).

mvollrath, 2020-12-17:
I can't recommend any personally, but I will say that the CBOR decoder needs a decent solution for tags because the rosbridge implementation uses the [draft typed array tags](https://tools.ietf.org/html/draft-ietf-cbor-array-tags-00) for performant handling of typed arrays (any numeric ROS array type other than byte string).  I would expect any decoder to at least give you data indicating which tag was applied and the tagged data as a byte string, which you then cast into the appropriate array type.  These tags are not in the spec yet so the decoder must support arbitrary tags.  At a glance the System decoder doesn't include these tags in the `CborTags` enum but I can't say for sure whether or not it matters.

The [roslibjs typed array decoder source](https://github.com/RobotWebTools/roslibjs/blob/develop/src/util/cborTypedArrayTags.js) may be a useful reference if you need to implement typed array support for your application.

amacneil, 2021-08-10:
Closing this issue since it is a discussion and not actionable. Per above, rosbridge has several serialization modes and some are lower overhead than others (JSON is slowest, CBOR-raw likely fastest). Outside of serialization, the overhead between websockets and TCP should be very low.

Please feel free to open another issue if there are specific performance issues you need help with, or open a thread in [ROS Discourse](https://discourse.ros.org/).

closed 2021-08-10 by amacneil