RobotWebTools/rosbridge_suite #547: Multiple client connection to rosbridge fails with different compression request

ller83, 2020-12-15:
I have a setup with two clients connecting to a rosbridge server.
One client is webviz and a second client is a web interface using rosjs connecting to rosbridge via websockets.

My question is if it is possible to connect multiple clients with different compression settings to the rosbridge server.

What I observed so far is, that I get an error message as soon as a second client with different compression settings as the first client is subribing to the same topic.

**Test 1**
1. connect to rosbridge with a client using no compression
2. connect to rosbridge with a second client using 'cbor-raw' compression (webviz)

Error message in rosbridge:
[MultiSubscriber.callback]: Exception calling subscribe callback: 'DiagnosticArray' object has no attribute '_buff'

**Test 2**
1. connect to rosbridge with a second client using 'cbor-raw' compression (webviz)
2. connect to rosbridge with a client using no compression

Error message in rosbridge:
 [MultiSubscriber.callback]: Exception calling subscribe callback: 'AnyMsg' object has no attribute '_slot_types'

## Specifications

  - ROS Version (noetic):
  - OS Version (`focal`):
  - Rosbridge Version (0.11.10):
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):
  - rosbridge_server Version (0.11.10)

mvollrath, 2020-12-16:
This may be caused by the server's caching of serialized data to prevent re-serializing the same data for each receiving client.

I would look at [subscribe capability logic for different compression types](https://github.com/RobotWebTools/rosbridge_suite/blob/5627cd6e96aa4be2502136c960f34f73d4d9c9a1/rosbridge_library/src/rosbridge_library/capabilities/subscribe.py#L323-L340) and [the OutgoingMessage caching object](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/internal/outgoing_message.py).

achim-k, 2022-10-18:
Reopening the issue as it is still present.

`cbor-raw` requires the internal `rospy.Subscriber` to subscribe to the topic with the generic [AnyMsg](http://docs.ros.org/en/melodic/api/rospy/html/rospy.msg.AnyMsg-class.html) as the message type. All other compressions require the specific message type, so depending on which message type was used to subscribe, the other compressions might not work for that topic.

https://github.com/RobotWebTools/rosbridge_suite/blob/216143f8dd59dab3c2d4104fd058ecad6519da6b/rosbridge_library/src/rosbridge_library/capabilities/subscribe.py#L120-L121

As a workaround, you can run two separate `rosbridge_server` which would basically give you two websockets to connect to: One for cbor-raw and one for the other compressions.