RobotWebTools/rosbridge_suite #549: Here`s question about the link between ROS and ROS2

wnalsqja1, 2021-01-12:
Hello I`m a college stuent who wants to build a ros-based platform.

it`s an approximate question but, i have a question.

I want ros and ros2 to be linked

To do that, i understand that i have to use rosbridge.
Is this right? Please give me a link if you have any information to refer to.

The reasons for the connection are as follows.
i want to send camera information from ros to ros2 and sensor data from ros2 to ros1 

I`d appreciate your answer.

thank you.

reinzor, 2021-01-12:
Please take a look at https://github.com/ros2/ros1_bridge

wnalsqja1, 2021-01-13:
thank you for your cooperation.

amacneil, 2021-08-10:
For anyone else coming across this issue:

- [ros1_bridge](https://github.com/ros2/ros1_bridge) is a project to bridge ROS 1 and ROS 2
- [rosbridge](https://github.com/RobotWebTools/rosbridge_suite) (this repo) is a project to connect to ROS over a websocket connection

There is no connection between these projects except for having "bridge" in the name.

closed 2021-08-10 by amacneil