RobotWebTools/rosbridge_suite #559: Rosbridge lagging massively when a occupancy_grid message is passed through

masterX244, 2021-02-15:
## Expected Behavior
High Frequency Messages not getting halted until map message is sent.

## Actual Behavior
Multiple seconds of pause before the map message is sent and then all High Frequency Messages are getting sent in one block.

## Steps to Reproduce the Problem

  1. Start slam_gmapping
  2. Start rosbridge
  3. Connect a listener on a topic that is sent in a high interval (the case where i spotted the bug had one sending current position every 0,2 seconds) with a queue size >0
  4. Connect a listener to the /map topic with a queue size >0

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): Kinetic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): DISTRIB_CODENAME=xenial
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`):0.11.12
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):16.0.0

mvollrath, 2021-02-16:
Using `cbor` compression for `OccupancyGrid` topics should help quite a lot.  JSON+b64 encoding of large binary blobs is very expensive on both ends.

kfirner, 2021-08-18:
Wtf homies... Did everybody get a 20 banger email on this?!?

amacneil, 2021-08-18:
See https://github.com/RobotWebTools/rosbridge_suite/pull/590: This project has a large number of low quality issues (some are not even reproducible problems, just open ended questions/discussions). Once stale issues are closed it will make it easier to focus on fixing high priority items.

amacneil, 2022-02-15:
Closing because OP has not responded to suggestions above to use cbor encoding.

closed 2022-02-15 by amacneil