RobotWebTools/rosbridge_suite #564: When I subscribed to the image topic, the image size was 1920 * 1080, and **the delay was very serious, about 1 minute.**

KainanSu, 2021-03-14:
I use rosbridge_suite to communicate with Android phones in Ubuntu.

## Expected Behavior
I am trying to use rosbridge_suite to subscribe the Image topic pubish by ros

## Actual Behavior
When I subscribed to the image topic, the image size was 1920 * 1080, and **the delay was very serious, about 1 minute.**

However, the topics published by the mobile terminal to Ubuntu do not delay, only the topics subscribed by the mobile terminal lag behind, and all the subscription topics lag behind.

If the image size is changed to 80 * 60, there is no such problem.

**Please help me, please help me**. I searched on the Internet for quite a long time, but I couldn't find a solution.

I tried to imitate this method and set the throttle_ rate = 20, but it doesn't work.
https://github.com/gramaziokohler/roslibpy/issues/16

## Specifications

  - ROS Version (`echo $ROS_DISTRO`): melodic
  - OS Version (`grep DISTRIB_CODENAME /etc/lsb-release`): DISTRIB_CODENAME=bionic
  - Rosbridge Version (`roscat rosbridge_server package.xml | grep '<version>'`): <version>0.11.13</version>
  - Twisted Version (`python -c 'import twisted; print twisted.version'`):[Twisted, version 17.9.0]

mvollrath, 2021-03-15:
1080p is a lot of data and the default is to encode it as base64 in JSON which is slow on both ends.  Use `cbor` compression type for large binary blobs like image data.

KainanSu, 2021-03-16:
Thank a lot, Movllrath.

 I changed to subscribe `compressedImage`, which reduced the bandwidth from 10MB to 1MB.

It compacts the image into `JPG `format, and on Android he can easily convert JPG to Bitmap and display it on the UI using only the `BinaryFactory`.

closed 2021-08-13 by amacneil